lora-rs/lora-rs #1: Status of crate?

Dash83, 2017-11-24:
Hello,

  I was wondering if you'll continue with the development of this crate. I'll soon be working with some Lora radios and I want to do it in Rust. If you are not continuing this library, I might take over myself. Can we discuss the direction of the crate somewhere?

Cheers.

ivajloip, 2017-11-30:
Hello,

Thank you very much for the interest in the project! Yes, my intention is to have something working by the end of the year. I will try to improve the organization of the project so that contributing is easier. For now ideas for things that need to be finished can be found [here][1]. I believe we should be able to use the newly-created gitter *community* [here][2] (if you have any trouble with that, please let me know) to discuss the development.

If you want to help me with the work, we can discuss what might be the most appropriate things to change in the code and have some pull requests for them. 

Cheers!

P.S: Sorry for the slight delay, but I was on vacation those last couple of days.

[1]: https://github.com/ivajloip/rust-lorawan#next-steps
[2]: https://gitter.im/rust-lorawan/

ivajloip, 2019-05-25:
Closing due to inactivity.

closed 2019-05-25 by ivajloip