lora-rs/lora-rs #6: RUSTSEC-2016-0005

IGBC, 2020-01-30:
https://rustsec.org/advisories/RUSTSEC-2016-0005.html

please switch to modern alternative AES library, without a std lib dependency.

ivajloip, 2020-01-31:
Thank you for pointing this out! I will look into it in the nearest future (hopefully within 1-2 weeks). If that would be too much for you, feel free to start a discussion on the alternatives and once we are on the same page, please consider creating a pull request :smiley:

IGBC, 2020-01-31:
Don't bother I forked it and rewrote it last night.

Once its cleaned up we can talk about future

lthiery, 2020-02-17:
Very interested in this nostd PR @IGBC

IGBC, 2020-02-24:
the library was half rewritten in 1 night, It needs a lot of cleaning up and some of the tests are broken / commented out. 

Bringing it up to a usable state is not something I can do by myself so if anyone is interested in helping please let me know.

ivajloip, 2020-02-25:
Hello @lthiery, 

Thank you for your interest in the library and for your feedback about something that you would like to have next.

@IGBC, I saw you had made 2 commits around the time you created this issue. Is there more code that you have written around this or what you have published is all that was needed for you? If you don't have the time to continue working on this now, I will try to find some time this or next weekend to separate the update of the crypto lib from the part that removes the STD dependency and work to fix the tests (probably in that order). I am also planning on seeing which parts of the library might not be necessary in all cases (to_string comes to mind) through features.

IGBC, 2020-02-26:
No I believe that's everything I have done so far. For the forseeable future I don't really have any time to continue working on this, especially as I do not have access to working hardware anymore. So if you want to pick up the batton go for it.

ivajloip, 2020-03-01:
Thanks :slightly_smiling_face: I just pushed a new version with the crypto library updated with commit 168bca8 (available with version 0.4.2). I believe this is resolving the current issue and I created a separate issue #7 to track the request to make the library slimmer and not dependent on std. Please reopen if you believe there is anything more that needs to be done for the current issue.

closed 2020-03-01 by ivajloip