lora-rs/lora-rs #7: Remove dependency on std

ivajloip, 2020-03-01:
Please refer to #6 for more information.

ivajloip, 2020-03-08:
Hi,

@lthiery, @IGBC I was looking at options to remove std from the core part of the library while still providing `to_string` as optional part of the library - if people want to use it (and similarly to move some other parts of the library to separate modules). What I believe should work is extract those to separate modules and do the implementation of `to_string`, etc in such additional modules. My thinking is that this way only those additional modules will depend to std and the core part of the library will not. It seems to me that this will solve your issue as you would depend only on the core part. As I do not have tons of experience with rust modules and embedded implementations for now, could you confirm that this plan sounds good to you?

Thanks in advance! If you or someone else provides feedback that this will be a working solution, I should have something testable some time next week.

IGBC, 2020-03-08:
You could also gate them behind feature flags, this is done by a lot of other libraries who aim for std/no-std compatibility.

IMO there's no right or wrong way to do it as long as how to use it in both configurations is clearly documented with examples.

_(emailed in from a phone, exuse my brevity)_

ivajloip, 2020-03-14:
Thank you very much for your reply! I ended up having a mixture of separate module + flags for now. @lthiery, please let me know if you face any issues with this solution. The version is currently released as `0.5.0-rc0`.

lthiery, 2020-03-14:
Awesome! I have some Rust bindings for the Semtech SX12xx drivers that I hope to connect to this soon. Thanks @ivajloip

closed 2020-04-12 by ivajloip