lora-rs/lora-rs #9: Fragmenting into Types more

lthiery, 2020-03-19:
Hi @ivajloip 

Thanks for putting this project together. I got to spend a full day with it today trying to actually do something and there's something that I think I can help improving but I wanted to get your opinion on it first.

I think `GenericPhyPayload` could be more of a builder type or it might even just be a public function that returns some MType. The reason is I think it's best to avoid functions like `decrypted_payload` that only apply to certain MTypes; if you transformed the data into a type you could use that to make sure the inapplicable function doesn't get called.

In addition, you may want to consider enriching either expanding the MTypes to have an Encrypted/Decrypted variant when applicable. Or they could have a payload field which can is an enum of Encrypted([u8]) vs Decrypted([u8]). Then you could take an encrypted packet and either transform the type itself from Encrypted to Decrypted or match on the state of the subfield and do the transformation.

Anyway, these are just initial thoughts. Let me know what you think.

Best,
Louis

ivajloip, 2020-03-19:
Hi @lthiery 

On first read it makes sense, we can have a function that returns an enum type and all of the enum values have different types with different methods. I also like the idea of Encrypted vs Decrypted - I remember some time ago I was wondering what would be the best way to achieve such kind of behavior, but without changing GenericPhyPayload to function, that was non-obvious. This sounds as a very elegant solution, thank you!

I will try to spend some more time on this tonight and see if there are any obvious issues. Otherwise I can have a stab at it and we can discuss whether it is how you imagined it. How does that sound to you?

Best regards,
Ivaylo

lthiery, 2020-03-19:
That sounds great! As mentioned in https://github.com/ivajloip/rust-lorawan/pull/8, I would suggest making the MType variants that are "encrypted" vs "decrypted" instead of doing it in the payload inside of a struct.

So I think you could do something like this:
```
/// MType gives the possible message types of the PhyPayload.
#[derive(Debug, PartialEq)]
pub enum MType {
    JoinRequest,
    JoinAccept,
    UnconfirmedDataUp,
    UnconfirmedDataDown,
    ConfirmedDataUp,
    ConfirmedDataDown,
    RFU,
    Proprietary,
}

#[derive(Debug, PartialEq)]
pub enum JoinAccept {
    Encrypted,
    Decrypted,
}

#[derive(Debug, PartialEq)]
pub enum UnconfirmedDataUp {
    Encrypted,
    Decrypted,
}

...
```
As you can see, the sub-enums are quite repetitive... so maybe a macro helper for that.

lthiery, 2020-03-19:
And on second though, I think you decrypt functions might be kind of nice, since it'll be like this for the user:
```
let decryption_attempt = joint_accept.decrypt(&key);
match decryption_attempt {
   // packet was already decrypted or was now successfully decrypted
   JoinAccept::Decrypted => (),
   // MIC verification failed, so the original packet was returned still encrypted
   JoinAccept::Decrypted => (),
}
```

ivajloip, 2020-03-23:
Hi Louis,

Sorry for the slight delay. I took some time in order to try to implement something a bit less trivial with the library so that I could  get a better feeling about its shortcomings, which my current minimalistic programs did not catch. As I said, I agree with you that there is a need to be able to distinguish Encrypted and Decrypted payloads and not have methods related to data packets in the join packets for example. Where my vision does not fully coincide with the way I understand your suggestion is the idea to change the meaning of MType to include the encrypted/decrypted information. For one, it will not save me from having extra methods that should not be there (`calculate_data_mic`) and what is even more important to me, it will change the meaning from the LoRaWAN specification, which might make it more difficult for some people to understand. For those reasons, my current ideas is to have a `parse` function that return an enum that looks something like

```rust
enum PhyPayload<T> {
    JoinRequest(PhyJoinRequestPayload<T>),
    EncryptedJoinAccept(EncryptedPhyJoinAcceptPayload<T>),
    DecryptedJoinAccept(DecryptedPhyJoinAcceptPayload<T>),
    EncryptedData(EncryptedPhyDataPayload<T>),
    DecryptedData(DecryptedPhyDataPayload<T>),
}
```

Where the encrypted version have things like mic verification, decrypt and reading fhdr, fport for data packets. The decrypted version on the other hand will have some of the things that the encrypted version had (fhdr, fport for data packets) plus for example AppNonce for join accept, etc. 

If I am to make this change, it will break the API to some non-negligable extend, so I would need to spend some more time trying to make sure this is the API that I would like to have. How does that sound to you?

lthiery, 2020-03-23:
Hi Ivaylo,

I think the main thing you are saying is to avoid MType as the name of the enum? That sounds good to me and I agree keeping a name like that "pure" with the protocol is important.

However, I would urge for sub-enums as they make a convenient return type (in my opinion) for decryption attempts.

```rust
enum PhyPayload<T> {
    JoinRequest(JoinRequestPayload<T>),
    JoinAccept(JoinAccept<T>),
    Data(DataPayload<T>),
}

enum JoinAccept {
     Encrypted(EncryptedPhyJoinAcceptPayload<T>)
     Decrypted(DecryptedPhyJoinAcceptPayload<T>)
}
```

This allows,

```rust

impl JoinAccept {
    pub fn decrypt(self, key: AESKey) -> JoinAccept {
        ...
    }
}
```
In the decryption, MIC verification can determine whether or not it was a successful decryption and either return the original encrypted packet if it failed (JoinAccept::Encrypted) or the decrypted one if it succeeded.

One last note as I'm thinking about this is that you will probably want to keep the notion of DataUp and DataDown in the types themselves (rather than have something like a "direction" helper function). The reason is that if you don't, you'll end up with something like this:

```rust
impl Data {
    pub fn encrypt(self, key: AESKey) -> Data {
        match self.direction() {
           Direction::Up => // encrypt payload
           Direction::Down => // decrypt payload
        }
    }
}
```

With the above, I'd be afraid that the compiler might not be able to optimize out the crypto for decrypt when compiling for a end-device target. But Rust is very good at optimizations though so I may be terribly wrong on this.

ivajloip, 2020-04-07:
Hi Louis,

I think now I am mostly happy with the types that I have now in [improved-api branch][1]. Unless you or anyone else has concerns with the changes, I intend to merge this branch in master and have it tagged as `v0.5.0`.

[1]: https://github.com/ivajloip/rust-lorawan/tree/improved-api

closed 2020-04-12 by ivajloip