lora-rs/lora-rs #10: FOpts Bug

lthiery, 2020-03-23:
This will probably be solved with your current refactor, but I came across a bug today that can be exposed with the following unit test:

```rust
#[test]
fn test_mac_command_in_downlink() {
    let raw_data = [0x60, 0x5F, 0x3B, 0xD7, 0x4E, 0xA, 0x0, 0x0, 0x3, 0x0, 0x0,
    0x0, 0x70, 0x3, 0x0, 0xFF, 0x0, 0x30, 0xCD, 0xDB, 0x22, 0xEE];
    let packet = GenericPhyPayload::new(raw_data).unwrap();

    assert_eq!(packet.mhdr().mtype(), MType::UnconfirmedDataDown);

    match packet.mac_payload() {
        MacPayload::Data(data) => {
            let fhdr = data.fhdr();
            let fopts = fhdr.fopts().unwrap();

            // there should only be one fopts
            assert_eq!(fopts.len(), 1);

            match fopts[0] {
                MacCommand::LinkADRReq(_) => (),
                _ => assert!(false),
            }
        }
        _ => {
            assert!(false)
        }
    }
}
```

Instead of parsing as LinkADRReq, the MacCommand is parsed as LinkADRAns. This tells me that the MAC Command parsing isn't distinguishing behind a downlink and uplink packet properly. As you can see with the first assert, it is identified as a UnconfirmedDataDown yet LinkADRAns is a uplink MAC.

ivajloip, 2020-03-29:
Hi,

Thank you very much for finding and reporting this issue! Indeed, during my refactoring I noticed it as well and it was fixed in the refactored version (now available [here](https://github.com/ivajloip/rust-lorawan/tree/improved-api)). I have the bug fixed also in branch [v0.4-support](https://github.com/ivajloip/rust-lorawan/tree/v0.4-support) and I intend to release the fix shortly. Before I do that, I just wanted to confirm with you that while the test exposes a real issue, it appears that there is a slight error in the test. Judging by the 6th byte of the payload `0xA` there are 10 bytes of mac commands (FCtrl parsing) and ther are two LinkADRReq being sent - 2 LinkADRReq commands and two other bytes that are not mapped to mac commands in v1.0.3 of the LoRaWAN spec. As you correctly recognized  the test was confused about uplink vs downlink. It interpreted only one byte of mac command as payload of the supposed LinkADRAns and the next mac command id was 0 for the parser, which was not recognized and therefore the mac command processing stopped. 

Do you agree about this correction of the test?

lthiery, 2020-04-01:
I do agree - thanks for checking!

closed 2020-04-01 by lthiery