lora-rs/lora-rs #11: Implement lorawan 1.1 spec

ivajloip, 2020-04-05:
# Task list

- [ ] New header parsing and field renaming
- [ ] New key derivation helper functions
- [ ] New MAC Commands

# Other considerations

- Be able to support only 1.0, or both
- Reuse as much code as possible

lucasgranberg, 2023-02-23:
I started messing around in my own repo to see what would be needed for 1.1 or 1.0.4.
https://github.com/lucasgranberg/lorawan

This is just a proof of concept and doesn't work yet. I didn't do this as a PR or anything because I am not 100% of what I am doing. Also supporting the newer specs requires quite heavy re-structuring. ivajloip/rust-lorawan is very well written and I learned a ton from picking it apart.

My take-aways from this:

* **This is not so easy**
  It is not easy to get the matrix of version specific and  region specific stuff to work.
* **Mac and Radio (PhyRxTx) can be separated**
  The physical radio should not need to be bound to any of this only by a trait. Also the Radio should not be a part of the Mac but passed in any time it is needed (see: https://github.com/lucasgranberg/lorawan/blob/fe67ad836ebffdc6f5b2e2e91755ea72d4efed7f/src/mac/mod.rs#L26). For battery powered devices it is super helpful to have full access to the hardware so it can be put to sleep or dropped without having to implement any stuff like that in the Mac layer.
* **Region should not contain a state**
   only static methods. How data is treated differs from region to region but the data structures does not
* **The Mac layer should contain state and this differs from version to version.**
   I added Session, Credentials and Status structs that are version specific. This also ties in to answering to mac commands on the next transmission.
* **Some encoding stuff is version specific, most is not.**
   I tried separating the version specific stuff and the common stuff. That is why there are two folders called encoding.
   This is still super rough and most of the code is just copied from this repo so it mostly follows 1.0.2 I guess.
   Many mac commands are the same so the common one could be in src/encodinng/maccommands.rs and the version specific could be in src/mac/mac_1_0_4/encoding/maccommands.rs
* **Should end-device and gateway specific code be mixed?**
  I butchered the original encoding stuff pretty heavily to make it more device specific. Maybe the data structures could be separated more from the implementations? 
* **Class stuff (A,B,C) could be implemented as a separated layer**
  Even the A class needs some logic. This also ties in to the maccommands for max duty cycle. For now this can be the responsibillity of another repo or of the user but it should be possible to do with the traits exposed by this repo.

lucasgranberg, 2023-05-24:
I have made a lot of progress on this. I have a POC for how lorawan 1.0.4 could be implemented but the code quality is not that good in my projects. https://github.com/lucasgranberg/lorawan + https://github.com/lucasgranberg/lorawan-pilot

I am now in the process of working with a rust-lorawan/encoding fork and getting my changes over there. My first attempt became a huge commit that is probably too much to merge but it gives me an idea of how to split up the work.

Some of the stuff I have bumped in to:
1. Split up uplink and downlink messages (BC, easier to match without `_ => `)
2. Add ADR stuff to FHDR (easy)
3. Use Error enum instead of &str (own preferrence and easier error handling)

I created this omega commit on my own fork but decided against making it a PR. Better to do it in steps and re-do it after the formatting PR is done on a clean slate. Also probably better to discuss the changes before implementing them..

ivajloip, 2023-06-01:
Hi Lucas,

Thank you for taking a stab at this! I haven't recently checked the changes that you have made, but some time ago they seemed indeed too much to be easy to merge. I would like for us to discuss your experience and decide on smaller steps that can bring us to the implementation (feel free to reach out to me through email to discuss this). 

For example, enum errors are indeed something that makes sense and especially if the change doesn't create too much toil for lib users, I would say let's do it.

lucasgranberg, 2023-06-02:
Id love to have a chat, I can't seem to find your email address. Mine should be on my profile