lora-rs/lora-rs #26: Insufficient number of bytes error

lthiery, 2020-07-16:
I'm using the encoding module in the lorawan-sniffer library as you may know and I've come across a parsing error on some "real live bytes". It could very well be that these are from LoRaWAN v1.1, but I wanted to log a few of the errors here for reference:

```
error: insufficient number of bytes, with bytes: [65, 67, 77, 73, 58, 57, 48, 58, 50]
error: can not build EncryptedDataPayload from the provided data, with bytes: [117, 30, 247, 224, 134, 94, error: can not build EncryptedDataPayload from the provided data, with bytes: [150, 211, 198, 243, 46, 141, 87, 243, 227, 91, 114, 194, 94, 45, 95, 39, 0, 0, 18]
```

Anyway, I'll try to loop back and analyze this when I have a moment but I wanted to share the datapoints.

ivajloip, 2020-07-26:
Hi,

Thanks for reporting this. It seems to me that the first two payloads are indeed very short as indicated in the error message - if I am not mistaken, even for LoRaWAN 1.1 the shortest phy payload should be at least 12 bytes long (1b MHDR, 7b FHDR, 4b MIC). One possible explanation is that someone is using LoRa, but not LoRaWAN - in that case all bets for the length are off and there is no way to correctly detect the protocol in all cases. I am not sure right now the library verifies the major version, which could be useful for detecting this kind of situations. I will keep this issue open until I implement that :slightly_smiling_face: 

As for the last payload, the first byte should be represented as b'10010110', which indicates unsupported major bits (it should be 00, not 10). This suggests that the bytes are not a LoRaWAN payload for me and the parsing error is to be expected (I will look in more details which part of the code failed the parsing).

Please let me know if I can be of further assistance for this issue.

lthiery, 2020-07-27:
Hi Ivo,

Makes sense. And I think in summary it's fair to say there's no bug or issue here as if they are not valid 1.0.2 frames then they shouldn't be getting parsed!

That being said, as potential enhancements, it would be nice to put in specific parsing errors. As you mention, "Invalid major version". In addition, parsing the OptNeg field on the JoinAccept and rejecting if its set would also be good as that would correspond to LoRaWAN 1.1, for example.

Otherwise, "Insufficient number of bytes" is a big catch-all parsing error, which is functional but perhaps not as explicit as it could be.

--Louis

closed 2023-01-09 by lthiery