lora-rs/lora-rs #37: nostd support

huntc, 2021-04-27:
Thanks for providing the library!

I'm looking for codec support in `nostd` environments, specifically so that I can target embedded devices. Any plans or thoughts there?

huntc, 2021-04-27:
Actually, it looks as though there's been some consideration for `nostd` although here's what I get when I include the lorawan-encoding library:

```
error[E0463]: can't find crate for `std`
 --> /Users/huntc/.cargo/registry/src/github.com-1ecc6299db9ec823/lorawan-encoding-0.6.2/src/extra/mod.rs:9:1
  |
9 | pub extern crate std;
  | ^^^^^^^^^^^^^^^^^^^^^ can't find crate
  |
```

huntc, 2021-04-27:
Ah, it seems that features have to be disabled for `nostd`... would that be all?

```
lorawan-encoding = { version = "0.6.2", default-features = false }
```

ivajloip, 2021-04-27:
Hi,

If I recall correctly, you need to disable the `full` feature and it should work with `nostd`. Please let me know if you have troubles with that.

lthiery, 2021-04-27:
Yes, it as Ivo says, just disable full features. Here's an example Toml from an embedded projects:
https://github.com/lthiery/rust-sx12xx/blob/78c74f8258235c4702583a500449aed73646990c/Cargo.toml#L22

I need to update the master branch of rust-12xx to match the current lorawan-device stack here, but that PR shows 90% of the work to doing that.

closed 2021-04-27 by huntc