lora-rs/lora-rs #66: Release of lorawan-device and lorawan-encoding

lulf, 2022-04-20:
Could we create a release to crates.io of lorawan-device and lorawan-encoding? Depending on git revisions prevents releasing crates that depend on these.

I'm happy to do it for lorawan-device since it's never released before (and I can add you both as owners), but I think I'd need to be added as owner on crates.io for lorawan-encoding. 

@ivajloip @lthiery ^^

lthiery, 2022-04-20:
I'm all for that. Should we try to sync the version numbers though? Makes it easier when a client project is using both at the same time (avoid pulling in two versions of encoding).

lulf, 2022-04-20:
Agreed, maybe just 0.7.0 for both is a nice candidate? Would be nice to merge #61 first as well (and maybe I'll add support for that in the async device as well).

lulf, 2022-04-21:
I've added the async downlink confirmation support in #67

ivajloip, 2022-04-24:
Yes, I think it's about time we release both of them and I agree that having consistent versioning between the two will be beneficial. I am a bit split about the naming - should `lorawan-encoding` be pushed as `lorawan` as it was before ([here](https://crates.io/crates/lorawan), but I don't know if or how that can be done) or should we name it `lorawan-encoding` and just add a link to it in the `lorawan`. I don't think it's possible to add automatic redirection in crate.io. If you have ideas around that, I would really be interested to learn more. :slightly_smiling_face:

lulf, 2022-04-24:
I think it's fine to keep the lorawan crate name for the codec, and we can use lorawan-device for the device. 

Less hassle for existing users of the crate as well.

lthiery, 2022-04-24:
I'm on board with maintaining lorawan name for the codec. Seems simple enough that way.

ivajloip, 2022-04-24:
okay, I tried to add you both to the owners of lorawan crate. I think I was looking into it some time ago and got confused about how `encoding/Cargo.toml` should be modified, so if anyone has a good idea how to do it, please feel free to do so. Otherwise I can take a look at it :slightly_smiling_face:

lulf, 2022-04-25:
Is it ok to change it back to 'lorawan' in encoding/Cargo.toml?

lthiery, 2022-04-25:
> Is it ok to change it back to 'lorawan' in encoding/Cargo.toml?

Yea I think that's the way to go

lulf, 2022-04-26:
I've made a PR for the renaming #69 . It was a bit more work than I thought.

ivajloip, 2022-04-26:
It looks good to me, thanks a lot for doing this! :slightly_smiling_face:

lulf, 2022-04-26:
Crates released! I've added you both as owners to the lorawan-device crate

closed 2022-04-27 by lulf