lora-rs/lora-rs #71: MSB/LSB first?

HerrMuellerluedenscheid, 2022-05-09:
Hey, I'm trying to get an [rfm9x](https://learn.adafruit.com/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts) to work with [sx127x](https://docs.rs/radio-sx127x/latest/radio_sx127x/) library and `rust-lorawan`. I'm trying to send data to TTN.
The MISO data line looks reasonable and apparently my tiva board can talk to the lora chip.

So far the data doesn't make it to TTN. So, I'm trying to narrow down the problem. This is my `lorawan` configuraiton:
```
    lora.set_tx_power(20, 1).unwrap(); //Using PA_BOOST. See your board for correct pin.

    let dev_eui = &[0x70, 0xB3, 0xD5, 0x7E, 0xD0, 0x05, 0x01, 0x3A];
    let app_eui = &[0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00];  # this can be all 0, right?
    let app_key = [0x80, 0x5D,.. total of 16 hex copy poste from TTN ];

    let mut data = [0; 23];  # take from a test I think. Do I have to set data here?
    let mut phy = JoinRequestCreator::with_options(&mut data[..], DefaultFactory).unwrap();
    let key = AES128(app_key);
    phy.set_app_eui(app_eui)
        .set_dev_eui(dev_eui)
        .set_dev_nonce(&[0x28, 0x10]);             # this can be any nonce right? So I randomly chose these two byte.

    let payload = phy.build(&key).unwrap();
``` 

Apart from the inline comments I'm wonderin if `rust-lorawan` expects `dev_eui`, `app_eui` and `app_key` as msb or lsb first?

Best regards and thanks in advance!
Marius

lulf, 2022-05-09:
IIRC, for TTN you need to copy DEV EUI and app EUI as LSB, whereas app_key is MSB.

FWIW, I've got a working sx127x example here using lorawan + lorawan-device (async): https://github.com/embassy-rs/embassy/blob/master/examples/stm32l0/src/bin/lorawan.rs

If you have access to your own gateway, you can see successful/failed join attempts in the TTN gateway log.

HerrMuellerluedenscheid, 2022-05-18:
Thanks for the info!

closed 2022-05-18 by HerrMuellerluedenscheid