lora-rs/lora-rs #74: P2P mode?

rcoder, 2022-07-02:
Apologies for opening an issue for what could be a mostly code-free discussion, but: based on my read of the `async_device` module it looks at least plausible that I could (mis)use the `Device#send_recv` method to do direct P2P comms between devices using this driver. (I.e., I want to send and receive raw datagrams, without any LoRaWAN handshake, crypto, or IP overlay.)

Is that something that y'all would expect to more or less work? If so, would it be safe to build on it as a low-level API?

If there's work to be done to clean up the interface or increase the odds of P2P mode "just working" I'm happy to dig in. I just don't want to spend a substantial amount of time working with the direct send/recv path only to have the methods I used get removed or refactored away soon.

lthiery, 2022-07-18:
If you want to use LoRa directly, then you should use the sx12xx drivers directly and ignore this library entirely.

I have a crate here (that probably needs some updating): https://github.com/lthiery/rust-sx12xx

It simply wraps the Semtech driver and brings it into Rust.

Otherwise, I think @lulf has a pure Rust implementation as well, but I can't find it off-hand right now.

closed 2023-01-09 by lthiery

lulf, 2023-01-09:
A few radio drivers here https://github.com/embassy-rs/embassy/tree/master/embassy-lora/src

Also this one is being worked on as a generic HAL-independent radio driver https://github.com/ceekdee/lora