lora-rs/lora-rs #77: Async device timing

CBJamo, 2022-07-18:
In [rx_with_timeout](https://github.com/ivajloip/rust-lorawan/blob/master/device/src/async_device/mod.rs#L338), the delay times are being calculated as if they were offsets from some zero time, rather than a duration for delay_ms. This makes both windows too long, and RX2 late. For example, with a join request, RX1 will open at 5s from tx done, but will stay open for the window + 5s, then RX2 will open 6s after RX1 closes, and be open for the window + 6s.

I hacked around and have working code, but frankly it's ugly and the math is convoluted. The offsets math is much easier to understand. Because of that I think it would be better to extend the timers to also accept a start and offset time. In my case that's trivial as I'm in userland and the [timer I'm using](https://docs.rs/async-io/1.7.0/async_io/struct.Timer.html#) already supports both after/duration and at/instant timers. I'm not sure how hard this change would be to make in the no_std/embedded world. If that's too big a task I can clean up my code and open a PR.

timokroeger, 2022-07-18:
Pushed some workaround for the late RX2 window which I had laying around as intermediate solution.
I also agree that a "deadline" timer API would make the code nicer than the current "delay" timer API.

CBJamo, 2022-08-18:
Closed by #79

closed 2022-08-18 by CBJamo