lora-rs/lora-rs #82: EU868 RX2 window datarate on TTN

CBJamo, 2022-09-08:
While reviewing #81 and preparing a followup pr for IN865 and AS923 I found out that TTN uses a non-standard datarate for RX2 in the EU[1]. The TTN docs claim that the OTAA process should configure this, but I don't believe that's being handled at the moment. Missing RX2 windows probably isn't a huge issue (I don't think I've ever seen a RX2 downlink, though I'm not in the EU), and I think this can wait till someone has time to refactor the region code.

[1] https://www.thethingsnetwork.org/docs/lorawan/frequency-plans/

ivajloip, 2022-09-10:
Hi Caleb,

Thank you for bringing this to our attention! My guess is that they are using RX2 with SF9BW125 after the join and in that case indeed devices are supposed to handle that when using OTAA (it's an option in the accept message - I will have to check if that is currently implemented for the device). If they are using the same channel for the join request, then it doesn't sound aligned with my understanding of the specification, but it's been a while since I last read it :smile:

lthiery, 2023-09-01:
I believe this is simply related to handling `RXParamSetupReq` MAC Comand from the LNS (and the `RXParamSetupAns` response).

[image]

closed 2025-06-06 by plaes