lora-rs/lora-rs #93: Use a specific subband when the network's subband is unknown

jbeaurivage, 2022-11-21:
Is there a way to make a device use a specific subband when the network's (or gateway's) operating subband is unknown before joining? For example in the US915 region, if no subband is specified (`US915::new()`), then the device will randomly hop across the 64 available channels. 

Here is an interesting use case: when first attempting to join, the device tries an OTAA join request on random channels until one is accepted. Is there a way to then restrict the device to then only use the subband on which it successfully joined? Or at least manually change the region/subband of a created `Device`?

Forgive my ignorance; I'm pretty new to LoRaWAN. Is it common for networks to all operate on the same subband, or are gateways allowed to operate on different subbands?

lthiery, 2022-11-21:
This kind of behavior is possible and I believe it would remain within spec. 

I previously wanted to implement it and would phrase it: "Upon successful join, send the first data frame on the same sub-band as the successful join". Then, I assume you're hoping for your first data frame response (ie: the downlink) to contain the ADR command with the channel mask that provides the appropriate channel restrictions for your network.

However, you should know, that if your LNS implements the newer CFList for US Regions (I think introduced in 1.0.4), you can leverage that instead; that is ultimately why I never implemented the approach above. The CFList is contained in the JoinResponse and contains the ChannelMask 

I never got around to absorbing the ChannelMask from the CFList appropriately, but [some of the pieces are there](https://github.com/ivajloip/rust-lorawan/blob/44c1f4bd6d173e7c4cd767acca470d535756c886/encoding/src/parser.rs#L470). I think the next step would be doing something with it [here](https://github.com/ivajloip/rust-lorawan/blob/master/device/src/region/us915/mod.rs#L35).

Also, just know that the current "random channel selection logic" doesn't map onto the ChannelMask stuff very well so there'd likely need to be some changes there. The current implementation is very "sub-band" focused when, in fact, there's no notion of sub-bands in the spec and it's just 72 channels and users happen to discuss things in terms of sub-bands. That is to say, the ChannelMask disables arbitrary channels with no notion of sub-band.

And just for the sake of clarity, the wildly out of spec [US::set_subband(&mut self, subband: u8)](https://github.com/ivajloip/rust-lorawan/blob/master/device/src/region/us915/mod.rs#L46) is available when initializing the stack; I admit that I make use of it but it is very inappropriate for anything remotely close to production. The major downfall is that you're basically out of luck if you ever want the device to work on a different sub-band.

jbeaurivage, 2022-11-22:
Thanks for the super detailer writeup. The JoinResponse CFList approach does seem like it makes sense to me. I might be interested in having a go at implementing it when I find some time, and I'm more familiar with this codebase. Some follow-up questions:

* As described in #91, it seems like the current ChannelMask implementation may be wrong; is this something that would prevent such an implementation from working correctly?
* If the newer JoinResponse CFList is implemented, does this mean that this stack would need to conform to the whole of LoRaWAN 1.0.4? Or can the LNS be made to operate spec version 1.0.4, and we may choose piecemeal which newer features we want to implement? Currently it seems like the project conforms to 1.0.2 (and I'm just saying this because I got that info from the README).

lthiery, 2022-12-26:
Hi @jbeaurivage - sorry for the delay in my response.

> As described in https://github.com/ivajloip/rust-lorawan/issues/91, it seems like the current ChannelMask implementation may be wrong; is this something that would prevent such an implementation from working correctly?

Yes, I believe so, yes. I'll try to take a look at that issue.

> If the newer JoinResponse CFList is implemented, does this mean that this stack would need to conform to the whole of LoRaWAN 1.0.4? Or can the LNS be made to operate spec version 1.0.4, and we may [...]

The whole device stack is experimental  and is only a partial LoRaWAN implementation as is. In my opinion, it's fine to selectively implement 1.0.4 features and I don't foresee any performance downsides or LNS errors in doing so, but I could be wrong about that.

lthiery, 2022-12-27:
@jbeaurivage I took a look at things #91 and wrote a response. I created #95 which should make leveraging ChannelMask in the fixed channel plans CFList more straightforward.

lthiery, 2022-12-29:
@jbeaurivage you might want to check out and/or test #96 

Leveraging CfList in JoinAccept is supported and the channel selection logic is spec compliant. From here, "biasing" the stack to start with a specific subband while remaining compliant seems possible.

PR #110 (2023-10-21): Improve channel selection mechanism for join requests using fixed channel regional plans
For US915 and AU915 regions, it would be beneficial to have a way of "biasing" the channels for join requests. This would enable quicker joins, instead of having to wait for a random channel selection to coincide with the network's enabled channels. This PR provides a way to achieve that.

Also fixes a bug in the datarate selection: If channel 63 is chosen, DR4 would've been selected. It is also very easy to implement the TODO to also pick from all 72 available channels:
```diff
- (random_number % 64) as u [... cut]
closed 2023-10-21 by lthiery