lora-rs/lora-rs #115: Avoid RX1 cancellation during RF reception

lthiery, 2023-08-31:
```
3.3.1 Receiver activity during receive windows
If a preamble is detected during one of the receive windows, the radio receiver SHOULD stay
active until the downlink frame is demodulated. If a frame was detected and subsequently
demodulated during the first receive window, and the frame was intended for this end-device
after address and MIC (message integrity code) checks, the end-device SHALL NOT open
the second receive window.
```

The code [here](https://github.com/ivajloip/rust-lorawan/blob/7f7ceef024991d0f068b59cff715897b021e8521/device/src/state_machines/session.rs#L502C18-L502C18) should handle `radio::Response::Rxing` appropriately.

lthiery, 2023-10-12:
#142 is addressing this for the async stack

plaes, 2024-01-18:
```
... and the frame was intended for this end-device after address and MIC (message integrity code) checks, the end-device SHALL NOT open the second receive window.
```
Although async_device now supports packet reception overflowing the RX1/RX2 windows, it doesn't check whether the frame was intended to this device.

lthiery, 2024-01-23:
> [The async_device stack] doesn't handle the case where the frame was not intended to this device and skips listening during RX2.

Actually, [I think it is well handled](https://github.com/lora-rs/lora-rs/blob/258ce4bbcb2e71075e52db91f4600f43641059b3/lorawan-device/src/async_device/mod.rs#L403). [The mac layer returns NoUpdate](https://github.com/lora-rs/lora-rs/blob/main/lorawan-device/src/mac/session.rs#L162) unless it's successful basically.

plaes, 2024-01-23:
> > [The async_device stack] doesn't handle the case where the frame was not intended to this device and skips listening during RX2.
> 
> Actually, [I think it is well handled](https://github.com/lora-rs/lora-rs/blob/258ce4bbcb2e71075e52db91f4600f43641059b3/lorawan-device/src/async_device/mod.rs#L403). [The mac layer returns NoUpdate](https://github.com/lora-rs/lora-rs/blob/main/lorawan-device/src/mac/session.rs#L162) unless it's successful basically.
> 
> If you agree @plaes, feel free to close this issue.

Yeah, let's close this one as async case was handled by #208 

And then create another issue for following flow (if needed):
1. Listen for RX1 and receive something...
2. Hook into HeaderValid and check if MIC/Header is for our device...
3. ...if not, skip to RX2 if it's still possible timing-wise.

closed 2024-01-23 by plaes