lora-rs/lora-rs #116: stm32wl55

tcpipchip, 2023-09-03:
Plans to au915 ?

lthiery, 2023-09-04:
What do you mean? AU915 is provided by this repo: https://github.com/ivajloip/rust-lorawan/blob/7f7ceef024991d0f068b59cff715897b021e8521/device/src/region/mod.rs#L19

tcpipchip, 2023-09-09:
nice, i finished the RUST blog about LSM110A, an stm32wl55 module.
https://lom204-cli-wisol.blogspot.com/2023/09/lsm110a-blink-led-com-rust.html
next step is test your project

lthiery, 2023-09-10:
It looks like you're not using an async runtime yet (like embassy).

So you can either look into that or you try to use the no_async device and call it with the appropriate events. [The example here](https://github.com/lthiery/rust-sx12xx/blob/master/examples/stm32l0x2/main.rs) shows the usage but it still uses a framework for managing/prioritizing the interrupts (the framework is called RTIC... which I think has some async features now? I'm not up to date).

tcpipchip, 2023-09-10:
Yes!

well, I will study your code and see how to port to stm32wl55

thanks

tcpipchip, 2023-09-10:
hi, could you help me to make work on E5 ? Then late i port to LSM110A a make a good blog

lthiery, 2023-09-11:
I can try to help. What are you trying to do exactly on the E5? Do you have a repo somewhere?

tcpipchip, 2023-09-11:
@lthiery 

Wait, i still studyng EMBASSY!

I got sucess on BLINK TEST  now!

https://lom204-cli-wisol.blogspot.com/2023/09/lsm110a-blink-led-com-rust-com-o-embassy.html

going to lorawan test

tcpipchip, 2023-09-11:
```
//! This example runs on a STM32WL board, which has a builtin Semtech Sx1262 radio.
//! It demonstrates LoRaWAN join functionality.
#![no_std]
#![no_main]
#![macro_use]
#![feature(type_alias_impl_trait, async_fn_in_trait)]

use defmt::println;
use defmt::info;
use embassy_executor::Spawner;
use embassy_lora::iv::{InterruptHandler, Stm32wlInterfaceVariant};
use embassy_lora::LoraTimer;
use embassy_stm32::gpio::{Level, Output, Pin, Speed};
use embassy_stm32::rng::{self, Rng};
use embassy_stm32::spi::Spi;
use embassy_stm32::{bind_interrupts, pac, peripherals};
use embassy_time::Delay;
use lora_phy::mod_params::*;
use lora_phy::sx1261_2::SX1261_2;
use lora_phy::LoRa;
use lorawan::default_crypto::DefaultFactory as Crypto;
use lorawan_device::async_device::lora_radio::LoRaRadio;
use lorawan_device::async_device::{region, Device, JoinMode};
use {defmt_rtt as _, panic_probe as _};

const LORAWAN_REGION: region::Region = region::Region::AU915; // warning: set this appropriately for the region

bind_interrupts!(struct Irqs{
    SUBGHZ_RADIO => InterruptHandler;
    RNG => rng::InterruptHandler<peripherals::RNG>;
});

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let mut config = embassy_stm32::Config::default();
    config.rcc.mux = embassy_stm32::rcc::ClockSrc::HSE32;
    config.rcc.rtc_mux = embassy_stm32::rcc::RtcClockSource::LSI;
    let p = embassy_stm32::init(config);

    pac::RCC.ccipr().modify(|w| w.set_rngsel(0b01));

    let spi = Spi::new_subghz(p.SUBGHZSPI, p.DMA1_CH1, p.DMA1_CH2);

    // Set CTRL1 and CTRL3 for high-power transmission, while CTRL2 acts as an RF switch between tx and rx
    let _ctrl1 = Output::new(p.PB12.degrade(), Level::Low, Speed::Low);
    let ctrl2 = Output::new(p.PC13.degrade(), Level::High, Speed::Low);
    let _ctrl3 = Output::new(p.PC3.degrade(), Level::High, Speed::High);
    let iv = Stm32wlInterfaceVariant::new(Irqs, None, Some(ctrl2)).unwrap();

    let mut delay = Delay;

    println!("********************************");
    info!("***********************************");
    info!("***********************************");

    let lora = {
        match LoRa::new(SX1261_2::new(BoardType::Stm32wlSx1262, spi, iv), true, &mut delay).await {
            Ok(l) => l,
            Err(err) => {
                info!("Radio error = {}", err);
                return;
            }
        }
    };
    let radio = LoRaRadio::new(lora);
    let region: region::Configuration = region::Configuration::new(LORAWAN_REGION);
    let mut device: Device<_, Crypto, _, _> = Device::new(region, radio, LoraTimer::new(), Rng::new(p.RNG, Irqs));

    defmt::info!("Joining LoRaWAN network");

    // TODO: Adjust the EUI and Keys according to your network credentials
    match device
        .join(&JoinMode::OTAA {
            deveui: [0x00,0x80,0xe1,0x15,0x05,0x1f,0xce,0xe4],
            appeui: [0, 0, 0, 0, 0, 0, 0, 0],
            appkey: [0x2a,0xed,0x51,0xb8,0x39,0xde,0x8d,0x83,0x07,0x56,0xb9,0x7d,0x2c,0x03,0xe8,0x2d],
        })
        .await
    {
        Ok(()) => defmt::info!("LoRaWAN network joined"),
        Err(err) => {
            info!("Radio error = {}", err);
            return;
        }
    };
}

```
Do you know for wich PORTA the info! redirecty the data by default on STM32WL55 ?

lthiery, 2023-09-11:
You might want to look at what @lucasgranberg has done here: https://github.com/ivajloip/rust-lorawan/issues/116#issuecomment-1714454383

He's using [a fork of the async rust-lorawan](https://github.com/lucasgranberg/lorawan) stuff (does not contain AU915 configuration) but you might find what you need there.

tcpipchip, 2023-09-11:
I will test it
Thanks

Enviado do meu iPhone

lthiery, 2023-10-12:
@tcpipchip i'm going to close this issue for now. please feel reach out on matrix for more support on your project if you need it 
https://matrix.to/#/#public-lora-wan-rs:matrix.org

closed 2023-10-12 by lthiery

tcpipchip, 2023-10-12:
ok, someday i will be back to the project...but i am happy that LoRa already working!