lora-rs/lora-rs #122: created a simple lora module here?

lthiery, 2023-09-04:
There has been previous discussion of how to encourage interoperability with other Rust Lora projects.

I think one thing that might help is if there was a simple crate to define the modulation's characteristics (ie: Bandwidth, CodingRate, and SpreadingFactor). We have it in here already, but perhaps we define a third-standalone crate that device can depend on? That would make it easy for projects to import those enums while avoiding any other cruft.

Potentially, different methods for calculating time-on-air could be included as well. Any thoughts?

lulf, 2023-09-04:
Agreed having a separate crate for this would be ideal, as I assume this could then be used by lora-phy to avoid all the conversions. And assuming this crate would not change frequently, it's easier to keep API and version compat sane.

lthiery, 2023-09-04:
Threw up a quick draft: https://github.com/ivajloip/rust-lorawan/pull/123/files

lthiery, 2023-09-04:
Note that ideally `lora/src/lora_phy.rs` would disappear as the hope would be that lora-phy adopts the lora crate.

ivajloip, 2023-09-06:
Yes, that makes sense to me as well. Ship it 🙂

ceekdee, 2023-09-06:
If lora-phy is to adopt this, please consider including all the bandwidths lora-phy supports:

pub enum Bandwidth {
    _7KHz,
    _10KHz,
    _15KHz,
    _20KHz,
    _31KHz,
    _41KHz,
    _62KHz,
    _125KHz,
    _250KHz,
    _500KHz,
}

LoRaWAN only uses the last three, but perhaps someone will want to do P2P using one or more of the others.

lthiery, 2023-09-10:
closed by #123

closed 2023-09-10 by lthiery