lora-rs/lora-rs #126: deprecate use of GenericArray

lthiery, 2023-09-11:
With const generics, it should be possible now. I think we use GenericArray because it impls Default though. I'm not sure what we need to use instead @lulf 

eg: 
```rust
let mut phy: DataPayloadCreator<[u8; 255], C> = DataPayloadCreator::default();
    |                                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `core::default::Default` is not implemented for `[u8; 255]`
```

plaes, 2024-09-18:
Firstly, `GenericArray` dependency is used both in our code and our dependencies, whenever `default-crypto` is used.

### Our unconditional usage
* `JoinAcceptCreator.encrypt_payload(...)`: https://github.com/lora-rs/lora-rs/blob/969ee2406d957954921d4be6fda69adac976534b/lorawan-encoding/src/creator.rs#L175-L179
* We use it in `securityhelpers::encrypt_frm_data_payload(...)`:
https://github.com/lora-rs/lora-rs/blob/969ee2406d957954921d4be6fda69adac976534b/lorawan-encoding/src/securityhelpers.rs#L61

* We re-export `generic_array` from securityhelpers: https://github.com/lora-rs/lora-rs/
blob/969ee2406d957954921d4be6fda69adac976534b/lorawan-encoding/src/securityhelpers.rs#L2

### Internal re-export usage
* `EncryptedJoinAcceptPayload.decrypt(...)`: https://github.com/lora-rs/lora-rs/blob/969ee2406d957954921d4be6fda69adac976534b/lorawan-encoding/src/parser.rs#L378-L379

* `Encrypter`, `Decrypter` and `Keys` traits: https://github.com/lora-rs/lora-rs/blob/969ee2406d957954921d4be6fda69adac976534b/lorawan-encoding/src/keys.rs#L187-L201

## Use by dependencies
Regarding dependencies, RustCrypto (from where our aes and cmac implementations are coming from) is trying to get rid of GenericArray dependency as well in favor of const generics - https://github.com/RustCrypto/traits/issues/1481

As an interim step they have introduced a [`hybrid_array` crate](https://docs.rs/hybrid-array/latest/hybrid_array) and have ported latest release candidate version to use it instead of GenericArray.

plaes, 2024-09-18:
I have a quick and dirty (with only minimal changes to get it to build) branch here that uses the pre-release crates: https://github.com/plaes/lora-rs/tree/hybrid-array

plaes, 2024-12-19:
Completed as per #309 by @lucasgranberg 

:100:  Thanks!

closed 2024-12-19 by plaes