lora-rs/lora-rs #134: Fix JoinMode Type

lthiery, 2023-09-17:
It's currently inconsistent in its use of AES128 type:
```rust
pub enum JoinMode {
    OTAA { deveui: [u8; 8], appeui: [u8; 8], appkey: [u8; 16] },
    ABP { newskey: AES128, appskey: AES128, devaddr: DevAddr<[u8; 4]> },
}
```
We probably want to make special types for DevEui and AppEui as well?

ivajloip, 2023-09-17:
Is [EUI64](https://github.com/ivajloip/rust-lorawan/blob/master/encoding/src/parser.rs#L912) something that will help here? Otherwise, sure, we can have separate types for those.

lthiery, 2023-09-17:
Yes, I do think EUI64 is appropriate for AppEui & DevEui. But we still might want to use a new type tho and do something like `struct AppEui(EUI64)`.

ivajloip, 2023-09-18:
sounds good! 🙂

lthiery, 2023-10-12:
#137 updates the device types to be more specific. It would still be good to update the encoding API to use these specific types as well. Work is being deferred for now as it should be paired with other breaking changes in the encoding library.

lthiery, 2023-12-02:
@ivajloip let me know how #177 looks to you

closed 2023-12-04 by lthiery