lora-rs/lora-rs #150: DBM values in Australia and America block usage of low-power ChipType::SX1261

ShakenCodes, 2023-10-16:
AU_DBM and US_DBM are set to 21, which is beyond the 15 dBm maximum when using ChipType::SX1261, including STM32WLxx configured for the Low-Power Amplifier. Need a way to pass in a different value, or change the default value when using ChipType::SX1261.

lthiery, 2023-10-16:
I think we need to copy @lucasgranberg idea and have there be a board trait for this. When setting the radio to a specific dBm, we can pass it thru a max filter that uses the "max board value". There's also a nuance that has to do with radio output and the antenna gain. For example, the 21 dBm includes antenna gain as far as I know.

lthiery, 2023-10-17:
I'm reading the specification and thinking about this... I'm seeing that @lucasgranberg allows the user to [implement max_eirp](https://github.com/lucasgranberg/lorawan/blob/main/src/device/mod.rs#L69-L70). My concern is that max_eirp is something that the LNS is allowed to set:
[image]

So I don't think that's quite the right thing here, because if you read the regional specifications, while they do allow some "out of band" configuration in some regions, in others they say it should otherwise default to X.

For example, the AU915 section
```
By default MaxEIRP is considered to be +30dBm. The Max EIRP can be modified by the
network server through the TxParamSetupReq MAC command and SHOULD be used by
both the end-device and the network server once TxParamSetupReq is acknowledged by
the device via TxParamSetupAns.
```

I'm wondering if we want there to be a trait like this
```rust 
trait BoardEirp {
    fn get_antenna_gain() -> u8 {
        0
    }
    fn max_radio_power() -> u8 {
        30
    }
}```

Then when we are targeting a specific EIRP (which is determined by TxPower + max EIRP), we first offset it by "get_antenna_gain" and then filter it through max(desired_output, B::max_radio_power()).

For example, TxPower 0 in the US would result in a desired 30 dBm EIRP (when max EIRP is at its default of 30). If your antenna gain is set to 2, we would want to get the radio to 28, but then we would get max_radio_power, and should we have 16, we would ask the radio to transmit at 16.

lucasgranberg, 2023-10-17:
The names (max_eirp, tx_power) are kind of misleading. Basically I wanted a way for the device to say what its capabilities are. Also taking in to account region and maccommand.
D = Device
R = Region

configruation.tx_power is set by LinkADRReq

The way I implemented it in Mac looks like this:
```rs
    /// Get the maximum EIRP for the end device.
    fn max_eirp<D: Device>() -> u8 {
        min(R::max_eirp(), D::max_eirp())
    }

    /// Get the transmission power based on the frame type.
    fn get_tx_pwr<D: Device>(frame: Frame, configuration: &Configuration) -> u8 {
        match frame {
            Frame::Join => Self::max_eirp::<D>(),
            Frame::Data => configuration.tx_power.unwrap_or(Self::max_eirp::<D>()),
        }
    }
```
In the region code for EU868 I have
```rs
    fn modify_dbm(tx_power: u8, cur_dbm: Option<u8>, max_eirp: u8) -> Result<Option<u8>, Error> {
        match tx_power {
            0..=7 => Ok(Some(max_eirp - (tx_power * 2))),
            15 => Ok(cur_dbm),
            _ => Err(Error::InvalidTxPower),
        }
    }
```

plaes, 2023-10-25:
Firstly, TX power depends on both chip abilities and region. (And then there are extra region-specific extra requirements, like CAD before TX in Japan?)

* Lora-Phy contains interfaces for configuring radio - It doesn't make sense to do any `Error::InvalidTxPower` there, instead I would prefer clamping tx power to be supported range of the chip. I already used [this approach in my WIP sx1272 driver](https://github.com/embassy-rs/lora-phy/pull/34/commits/e64229e9e37aaadb4588f059aaa17e8a91a91802#diff-6caa63a80d0bd788eb193999e8016f3d971049d661f8f64e61f756247e9122cfR110-R130).
* Regarding board Trait - I have a WIP PR for dropping the BoardType/ChipType global enum in favor of Config struct that could be expanded with the information about the antenna gain. (See [lora-phy #37](https://github.com/embassy-rs/lora-phy/pull/37) and [rust-lorawan #152](https://github.com/ivajloip/rust-lorawan/pull/152)).

And one extra thing that needs to be addressed in lora-phy side is output routing (which might warrant a redesign for the `prepare_for_tx` method in LoRa impl):
* sx1272 has two physical outputs (RFO and PA_BOOST) for TX which can be configured exclusively of eachother, but it's up to board designer whether both SFO and PA_BOOST are wired to antenna or one of them: [image]

* sx1276 (and ...7,8,9) - has even more complicated RF frontend due to its support for 3 different frequency bands
[image]

PR #159 (2023-11-09): extend PhyTxRx to consider antenna gain and max power
Resolves #150
closed 2023-11-09 by lthiery