lora-rs/lora-rs #183: error[E0706]: functions in traits cannot be declared `async`

David-OConnor, 2023-12-11:
Steps to reproduce:
Add this to `Cargo.toml`:
```toml
lorawan-device = "^0.11.0"
lora-modulation = "^0.1.3"
#lora-phy = "^2.1.2"
# crates release of `lora-phy` is no longer maintained, but the GH is.
#lora-phy = { git = "https://github.com/lora-rs/lora-rs/tree/main/lora-phy" }
lora-phy = { git = "https://github.com/lora-rs/lora-rs" }
```

On compile: 
```rust
error[E0706]: functions in traits cannot be declared `async`
  --> C:\Users\the_a\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embedded-hal-async-1.0.0-rc.2\src\delay.rs:22:5
   |
22 |     async fn delay_ms(&mut self, mut ms: u32) {
   |     -----^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |     |
   |     `async` because of this
   |
   = note: `async` trait functions are not currently supported
   = note: consider using the `async-trait` crate: https://crates.io/crates/async-trait
   = note: see issue #91611 <https://github.com/rust-lang/rust/issues/91611> for more information
```
Times ~50-100. Any ideas? Ty! I just got a board in with SX1262 wired to STM32, and am trying it!

plaes, 2023-12-12:
There was API change in embdedded-hal-async v1.0.0-rc.2:
* delay: Rename DelayUs to DelayNs

lucasgranberg, 2023-12-12:
async fn in trait or AFIT needs beta channel of rust to work. you can set it with a rust-toolchain.toml file

closed 2023-12-24 by David-OConnor

lthiery, 2023-12-24:
4 more days...