lora-rs/lora-rs #205: Example code for stm32wl only works in release mode

praguevara, 2024-01-13:
When I run the `lora_p2p_send` on a `NUCLEO-WL55JC1` in unoptimized mode, it doesn't work:

```
PS C:\Users\pragu\Documents\electronics\nucleo> cargo r
    Finished dev [unoptimized + debuginfo] target(s) in 2.41s
     Running `probe-rs run --chip STM32WLE5JCIx target\thumbv7em-none-eabi\debug\nucleo`
      Erasing ✔ [00:00:02] [##############################################################################################################################################################################################################################################] 142.00 KiB/142.00 KiB @ 60.51 KiB/s (eta 0s )
  Programming ✔ [00:00:03] [##############################################################################################################################################################################################################################################] 141.00 KiB/141.00 KiB @ 38.85 KiB/s (eta 0s )    Finished in 5.992s
0.000000 TRACE BDCR ok: 00008200
└─ embassy_stm32::rcc::bd::{impl#3}::init @ C:\Users\pragu\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embassy-stm32-0.1.0\src\fmt.rs:117 
0.000000 DEBUG rcc: Clocks { sys: Hertz(48000000), pclk1: Hertz(48000000), pclk1_tim: Hertz(48000000), pclk2: Hertz(48000000), pclk2_tim: Hertz(48000000), pclk3: Hertz(48000000), hclk1: Hertz(48000000), hclk2: Hertz(48000000), hclk3: Hertz(48000000), rtc: Some(Hertz(32000)) }
└─ embassy_stm32::rcc::set_freqs @ C:\Users\pragu\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embassy-stm32-0.1.0\src\fmt.rs:130 
```

Switching to release mode does:

```
PS C:\Users\pragu\Documents\electronics\nucleo> cargo r -r
    Finished release [optimized + debuginfo] target(s) in 1.49s
     Running `probe-rs run --chip STM32WLE5JCIx target\thumbv7em-none-eabi\release\nucleo`
      Erasing ✔ [00:00:00] [################################################################################################################################################################################################################################################] 48.00 KiB/48.00 KiB @ 59.29 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [################################################################################################################################################################################################################################################] 48.00 KiB/48.00 KiB @ 38.28 KiB/s (eta 0s )    Finished in 2.079s
0.104125 TRACE read: addr=[12], len=2, status=e2, buf=[00, 00]
└─ lora_phy::interface::{impl#0}::read_with_status::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:86  
0.104220 TRACE write: [02, 00, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:24  
0.104267 DEBUG process_irq satisfied: irq_flags = 0x0 in radio mode Transmit
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:882 
0.104319 DEBUG process_irq loop entered
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:857 
0.104407 TRACE read: addr=[12], len=2, status=e2, buf=[00, 00]
└─ lora_phy::interface::{impl#0}::read_with_status::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:86  
0.104501 TRACE write: [02, 00, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:24  
0.104549 DEBUG process_irq satisfied: irq_flags = 0x0 in radio mode Transmit
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:882
0.104600 DEBUG process_irq loop entered
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:857
0.104688 TRACE read: addr=[12], len=2, status=e2, buf=[00, 00]
└─ lora_phy::interface::{impl#0}::read_with_status::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:86
0.104783 TRACE write: [02, 00, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:24
0.104830 DEBUG process_irq satisfied: irq_flags = 0x0 in radio mode Transmit
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:882
0.104882 DEBUG process_irq loop entered
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:857
0.104970 TRACE read: addr=[12], len=2, status=e2, buf=[00, 00]
└─ lora_phy::interface::{impl#0}::read_with_status::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:86
0.105065 TRACE write: [02, 00, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:24
0.105112 DEBUG process_irq satisfied: irq_flags = 0x0 in radio mode Transmit
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:882
0.105164 DEBUG process_irq loop entered
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:857
0.105252 TRACE read: addr=[12], len=2, status=e2, buf=[00, 00]
└─ lora_phy::interface::{impl#0}::read_with_status::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:86
0.105346 TRACE write: [02, 00, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:24  
0.105394 DEBUG process_irq satisfied: irq_flags = 0x0 in radio mode Transmit
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:882
0.105445 DEBUG process_irq loop entered
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:857
0.105533 TRACE read: addr=[12], len=2, status=e2, buf=[00, 00]
└─ lora_phy::interface::{impl#0}::read_with_status::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:86
0.105628 TRACE write: [02, 00, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:24
0.105675 DEBUG process_irq satisfied: irq_flags = 0x0 in radio mode Transmit
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:882
0.105727 DEBUG process_irq loop entered
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:857
[... 246 lines cut ...]
0.127497 TRACE read: addr=[12], len=2, status=e2, buf=[00, 00]
└─ lora_phy::interface::{impl#0}::read_with_status::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:86
0.127592 TRACE write: [02, 00, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:24
0.127639 DEBUG process_irq satisfied: irq_flags = 0x0 in radio mode Transmit
└─ lora_phy::sx1261_2::{impl#1}::process_irq::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\sx1261_2\mod.rs:882
0.127691 DEBUG process_irq loop entered
[... cut ...]
0.128022 INFO  TX DONE
└─ nucleo::____embassy_main_task::{async_fn#0} @ src\main.rs:103
0.128072 TRACE write: [84, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ C:\Users\pragu\.cargo\git\checkouts\lora-rs-9ebdb91068f5935a\fabb564\lora-phy\src\interface.rs:24
0.130131 INFO  Sleep successful
└─ nucleo::____embassy_main_task::{async_fn#0} @ src\main.rs:112
```

I'm using the git source in my Cargo.toml, since the crates.io version isn't updated yet to the new update of embedded-hal-async.

plaes, 2024-01-19:
Yes, this is unfortunately a known issue with a known workaround - always build things in release mode (or tweak your profiles).

lucasgranberg, 2024-11-30:
When working on constrained devices release is often needed just to make it fit. Also some timings could change with debug vs release so it would be too hard to maintain something that has to work on both profiles.

The documentation could probably be improved here to avoid confusion when trying the examples. PRs welcome