lora-rs/lora-rs #218: RX2 downlink handling in case RX1 frame was detected but not intended for the end-device

plaes, 2024-01-23:
Handle following edge-case:
1. Listen for RX1 downlink and start receiving frame...
2. Hook into HeaderReceived (or any other possible mechanism) and validate MIC and check whether packet is intended for current end-device...
3. ...if not, make handle RX2 window as well (if possible timing wise).

lthiery, 2024-01-23:
[I think it is well handled](https://github.com/lora-rs/lora-rs/blob/258ce4bbcb2e71075e52db91f4600f43641059b3/lorawan-device/src/async_device/mod.rs#L403). [The mac layer returns NoUpdate](https://github.com/lora-rs/lora-rs/blob/main/lorawan-device/src/mac/session.rs#L162) unless it's successful basically.

Does this not satisfy the case your detailing?

lthiery, 2024-01-23:
Sorry, I should make an effort to rephrase:

1. We will proceed to RX2 as long as `fn rx_listen` returns `None`: https://github.com/lora-rs/lora-rs/blob/258ce4bbcb2e71075e52db91f4600f43641059b3/lorawan-device/src/async_device/mod.rs#L371
2. After processing a frame, the `fn rx_listen` returns `None` if the mac layer responds with: `NoUpdate`: https://github.com/lora-rs/lora-rs/blob/258ce4bbcb2e71075e52db91f4600f43641059b3/lorawan-device/src/async_device/mod.rs#L403
3. The mac layer responds with `NoUpdate` when it fails to parse the frame: https://github.com/lora-rs/lora-rs/blob/main/lorawan-device/src/mac/session.rs#L162

plaes, 2024-02-21:
> Sorry, I should make an effort to rephrase:
>
>     1. We will proceed to RX2 as long as `fn rx_listen` returns `None`: [...]

Ok, I'll take your word ont it.. :+1:

closed 2024-02-21 by plaes