lora-rs/lora-rs #226: embedded-hal-async-1.0.0-rc.1 `#![feature]` may not be used on the stable release channel

mjsir911, 2024-02-16:
Hi,

I'm a bit of a rust newbie, but I'm trying to figure out how to resolve an issue coming from one of your dependencies

```
error[E0554]: `#![feature]` may not be used on the stable release channel
 --> $HOME/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-async-1.0.0-rc.1/src/lib.rs:4:1
  |
4 | #![feature(async_fn_in_trait, impl_trait_projections)]
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  |
  = help: the feature `impl_trait_projections` has been stable since `1.74.0` and no longer requires an attribute to enable
```

I've tried adding `embedded-hal-async = "1.0.0-rc.2"` into my `Cargo.toml` because [they seem to have fixed this issue in that version](https://github.com/rust-embedded/embedded-hal/pull/534/files#diff-12ff19effb01592514fbd9b611c8e98309918524b9e794c1e3aea9c08b49e4a1), but cargo gives me a dependecy conflict:

```
error: failed to select a version for `embedded-hal-async`.
    ... required by package `meshtastic-rs v0.1.0 (.../meshtastic-rs)`
versions that meet the requirements `^1.0.0-rc.2` are: 1.0.0, 1.0.0-rc.3, 1.0.0-rc.2

all possible versions conflict with previously selected packages.

  previously selected package `embedded-hal-async v1.0.0-rc.1`
    ... which satisfies dependency `embedded-hal-async = "=1.0.0-rc.1"` of package `lora-phy v2.1.2`
    ... which satisfies dependency `lora-phy = "^2.1.2"` of package `meshtastic-rs v0.1.0 (.../meshtastic-rs)`

failed to select a version for `embedded-hal-async` which could resolve this conflict
```

I can't find where the `embedded-hal-async` is actually being pinned?? but [lib.rs](https://lib.rs/crates/lora-phy) reports the same, this is the only dep that is pinned to the rc.

I'm still learning this stuff, but would y'all be able to loosen restrictions a bit so that `embedded-hal-async-1.0.0-rc.2` (and potentially future rcs) can be pulled in?

mjsir911, 2024-02-16:
Okay this actually looks like the `lora-phy = "2.1.2"` / the page on lib.rs might be pointing to the deprecated repo https://github.com/embassy-rs/lora-phy? How am I supposed to pull in y'alls code in my `Cargo.toml`?

All of your examples are using relative paths & I don't really want to clone / vendor your code to refer to by path. Would be useful to have an example somewhere for how to actually pull in and use your library.

lthiery, 2024-02-16:
@mjsir911 we're still trying to wrap up some ideas before publishing lora-phy 3.0 on crates. But you should be able to replace the dependencies with the following:

```
lora-phy = { git = "https://github.com/lora-rs/lora-rs.git", branch = "main", features = ["lorawan-radio"] }
lorawan-device = { version = "0.8.0", default-features = false, features = ["embassy-time", "default-crypto", "defmt"] }
lorawan = { version = "0.12", default-features = false, features = ["default-crypto"] }
```

mjsir911, 2024-02-16:
Hi! Thanks for the swift response.

If this is on y'alls radar, I'll close out this issue. I appreciate the help.

I think I've narrowed down on this, too. Not sure if I'm using patches the proper way but:
```
[patch.crates-io]
lora-phy = { git = "https://github.com/lora-rs/lora-rs.git", tag="lorawan-v0.8.0" }
```

closed 2024-02-16 by mjsir911

lthiery, 2024-02-16:
That'll work fine but I would just specify 
```
[dependencies]
lora-phy = { git = "https://github.com/lora-rs/lora-rs.git", tag="lorawan-v0.8.0" }
```

No need to patch afaik. 

Concerning the Cargo.toml for the examples, I'm hesitant to _not_ set them relative because then we run the risk of examples being out of sync with the local crates... Something to chew on.

mjsir911, 2024-02-16:
> Concerning the Cargo.toml for the examples, I'm hesitant to not set them relative because then we run the risk of examples being out of sync with the local crates... Something to chew on.

That makes sense! Even just `lora-phy = { git = "https://github.com/lora-rs/lora-rs.git", tag="main" }` or something in the `README.md` would point users like me in the right direction ;-). Or maybe they'll run into this issue since presumably this error would come up on any rust version greater or equal to 1.74.0

plaes, 2024-02-16:
@mjsir911 `lora-phy = { git = "https://github.com/lora-rs/lora-rs.git", branch="main" }`