lora-rs/lora-rs #260: Examples not working on RAK4631

ionspin, 2024-07-02:
I've tried running the examples on RAK4631 and they fail to run with the following logs:
```
    Finished release [optimized + debuginfo] target(s) in 1.08s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabi/release/lora_p2p_receive`
      Erasing ✔ [00:00:01] [######################################################################################################################################################################################################################################################] 48.00 KiB/48.00 KiB @ 39.27 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [######################################################################################################################################################################################################################################################] 48.00 KiB/48.00 KiB @ 40.64 KiB/s (eta 0s )    Finished in 2.423s
0.040100 TRACE write: [c0, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:24
0.040191 TRACE write: [80, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:24
0.040313 TRACE write: [96, 01]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:24
0.040405 TRACE write: [9d, 01]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:24
0.040588 TRACE read: addr=[07], len=2, status=ff, buf=[ff, ff]
└─ lora_phy::interface::{impl#0}::read_with_status::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:86
0.040771 TRACE write: [97, 01, 00, 02, 80]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:24
0.040863 TRACE write: [89, 7f]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:24
0.040985 TRACE write: [8a, 01]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:24
0.041137 TRACE write: [0d, 07, 40, 14, 24]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:24
0.041290 TRACE write: [8f, 00, 00]
└─ lora_phy::interface::{impl#0}::write::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:24
0.041595 TRACE read: addr=[1d, 02, 9f, 00], len=9, data=[ff, ff, ff, ff, ff, ff, ff, ff, ff]
└─ lora_phy::interface::{impl#0}::read::{async_fn#0} @ /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/interface.rs:61
0.041687 ERROR panicked at /home/ionspin/Projects/Future/Rust2024/RustLora/rp2040Lora/lora-rs/lora-phy/src/sx126x/mod.rs:98:36:
index out of bounds: the len is 9 but the index is 9
└─ panic_probe::print_defmt::print @ /home/ionspin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.2/src/lib.rs:104
```

I've believe the cause to be configuration of sx1262 busy pin which is pulled down:
https://github.com/lora-rs/lora-rs/blob/b4b6741152b0bc599574eecff9a7ffc031ba1823/examples/nrf52840/src/bin/lora_p2p_receive.rs#L32

Changing the value of pulldown to `None` fixes the problem. 

I'm not completely sure why this is happening, I am guessing the pin is not floating and doesn't have to be pulled down, but I'm not an electronics expert.

lucasgranberg, 2024-07-05:
If I understand the documentation right the pull should be none or  Up. The example is not specifically made for rak4631 but I think this could actually be changed in code. I don't have a nrf52 based system but PRs welcome :+1: 

From the sx1261-2 documentation (page 53):
> The BUSY control line is used to indicate the status of the internal state machine. When the BUSY line is held low, it indicates that the internal state machine is in idle mode and that the radio is ready to accept a command from the host controller.

...

>In Sleep mode, the BUSY pin is held high through a 20 kΩ resistor and the BUSY line will go low as soon as the radio leaves the Sleep mode.

ionspin, 2024-07-05:
Thanks for the pointer from the documentation! I'll try to get a PR for nrf52840 examples over the weekend, since I have the boards.

PR #264 (2024-07-08): examples: nrf52840/RAK4630: Use Input(Pull::None) for busy pin.
From Sx1261-2 user manual:
> The BUSY control line is used to indicate the status of the internal state
> machine. When the BUSY line is held low, it indicates that the internal
> state machine is in idle mode and that the radio is ready to accept a
> command from the host controller.

Fixes #260.

Tested-by: Priit Laes <plaes@plaes.org>
commit fc20f7bb (2024-07-08): examples: nrf52840/RAK4630: Use Input(Pull::None) for busy pin.

From Sx1261-2 user manual:
> The BUSY control line is used to indicate the status of the internal state
> machine. When the BUSY line is held low, it indicates that the internal
> state machine is in idle mode and that the radio is ready to accept a
> command from the host controller.

Fixes #260.

Tested-by: Priit Laes <plaes@plaes.org>
closed 2024-07-08 by plaes