lora-rs/lora-rs #262: add sx1272 example

gilloud, 2024-07-04:
Will it be possible to add a sx1272 example? The integration of this board seems more complex for me due to the DIO interfaces integration. 

Thanks a lot

lthiery, 2024-07-05:
It should not diverge much from the stm32l0 example with the sx1276.

Do you have an easy to source board with SX1272 as a reference?

gilloud, 2024-07-14:
sx1272 use 4 dio as we can see it ( https://github.com/Lora-net/LoRaMac-node/blob/master/src/boards/SKiM980A/sx1272-board.c )
For me, the implementation si sigly different.

I have multiple board of this old model : https://github.com/wyres/hardware/blob/main/w_basev2/revB/W_BASE_V2.0_REVB_Schema.pdf STM32L151CC + SX1272 ).
From here  : https://www.tindie.com/products/wyld-things/lorawan-sensor-board/
for the EU868 region.

plaes, 2024-11-14:
You need to create a custom interfacevariant to listen events for multiple dio's, here's a partial example I am using for a custom device:
```rust
use embassy_futures::select::{select, Either};

pub struct Sx127xWithRxTx<WAIT> {
    dio0: WAIT,
    dio1: WAIT,
    _reset: WAIT,
}

impl<WAIT> Sx127xWithRxTx<WAIT>
where
    WAIT: Wait,
{
    pub fn new(dio0: WAIT, dio1: WAIT, _reset: WAIT) -> Result<Self, RadioError> {
        Ok(Self { dio0, dio1, _reset })
    }
}

impl<WAIT> InterfaceVariant for Sx127xWithRxTx<WAIT>
where
    WAIT: Wait,
{
    async fn await_irq(&mut self) -> Result<(), RadioError> {
        match select(self.dio0.wait_for_high(), self.dio1.wait_for_high()).await {
            Either::First(_) => {
                #[cfg(feature = "defmt")]
                defmt::info!("dio0");
            }
            Either::Second(_) => {
                #[cfg(feature = "defmt")]
                defmt::info!("dio1");
            }
        }
        Ok(())
    }
}
```