lora-rs/lora-rs #263: Is it possible to use CH582F chip with lora?

happytm, 2024-07-04:
It will be nice to have crate if possible to use low cost CH582F micro controller with built in BLE and USB host.

https://github.com/ch32-rs/ch58x-hal

Thanks.

lthiery, 2024-07-05:
It is possible but you will have to implement board support yourself combining the ch58x-hal with the appropriate Lora (sx12xx) driver. Do you have an easy to source board with the LoRa+CH582F ?

happytm, 2024-07-05:
Thank you for quick response.

I do have:
https://github.com/WeActStudio/WeActStudio.WCH-BLE-Core
https://www.cdebyte.com/products/E22-900M
https://www.cdebyte.com/products/E28-2G4M27S
https://www.cdebyte.com/products/E220-900T22S

And various other modules from E-byte.

Unfortunately I am new to Rust so I need your help with software, if possible.

Thanks.

Thanks.

closed 2025-01-06 by lthiery