lora-rs/lora-rs #282: Sx127x driver uses floating point math

plaes, 2024-09-01:
Floating point math is used in sx127x phy driver. This will cause issues on some companion chips ([for example esp32](https://github.com/esp-rs/esp-hal/issues/2050)).

lucasgranberg, 2024-09-02:
There is some in lorawan too:
https://github.com/lora-rs/lora-rs/blob/a25e970225f4976a7b71d7159f7c69b5b0b70860/lorawan-encoding/src/maccommands.rs#L711

Main culprits are simple conversions in a handful of places
https://github.com/lora-rs/lora-rs/blob/a25e970225f4976a7b71d7159f7c69b5b0b70860/lora-phy/src/sx127x/sx1276.rs#L145
https://github.com/lora-rs/lora-rs/blob/a25e970225f4976a7b71d7159f7c69b5b0b70860/lora-phy/src/sx127x/mod.rs#L307
https://github.com/lora-rs/lora-rs/blob/a25e970225f4976a7b71d7159f7c69b5b0b70860/lora-phy/src/sx127x/mod.rs#L386

narottamroyal, 2025-06-19:
I've thought about a couple of ways to resolve the max duty cycle calculation and wanted to see if there's a preferred approach/direction before making a pull request.

`max_duty_cycle_raw()` returns values in the range $[0, 15]$. Therefore, the divisor ranges from $[1, 2^{15}]$, and `max_duty_cycle()` returns values in the range $[2^{-15}, 1]$, or approximately $[0.000030518, 1]$.

## Scale the max duty cycle and return an unsigned integer

### Scaling by $2^{15}$

This approach retains full accuracy, but requires dividing the result by $2^{15}$, which isn't particularly ergonomic or easy to understand.

```rust
/// Value of the max duty cycle field as a portion of time scaled by 2^15.
pub fn max_duty_cycle(&self) -> u16 {
    1 << (15 - self.max_duty_cycle_raw())
}
```

### Scaling by $10^9$

Scaling by a power of 10 is easier to understand, but still isn't that ergonomic and results in a maximum error of $6.3 \times 10^{-10}$.

```rust
/// Value of the max duty cycle field as a portion of time scaled by 10^9.
pub fn max_duty_cycle(&self) -> u32 {
   1_000_000_000 / (1 << self.max_duty_cycle_raw())
}
```

## Compute the `f32` result without floating-point calculations

Looking at the raw bits for each duty cycle represented as an `f32`, we can observe that the value of the exponent decrements for each value, which makes sense since we're computing negative powers of 2:

```
Sign Exponent Mantissa                  Decimal value
   0 01111111 00000000000000000000000 = 1
   0 01111110 00000000000000000000000 = 0.5
   0 01111101 00000000000000000000000 = 0.25
   0 01111100 00000000000000000000000 = 0.125
   0 01111011 00000000000000000000000 = 0.0625
   0 01111010 00000000000000000000000 = 0.03125
   0 01111001 00000000000000000000000 = 0.015625
   0 01111000 00000000000000000000000 = 0.0078125
   0 01110111 00000000000000000000000 = 0.00390625
   0 01110110 00000000000000000000000 = 0.001953125
   0 01110101 00000000000000000000000 = 0.0009765625
   0 01110100 00000000000000000000000 = 0.00048828125
   0 01110011 00000000000000000000000 = 0.00024414063
   0 01110010 00000000000000000000000 = 0.00012207031
   0 01110001 00000000000000000000000 = 0.000061035156
   0 01110000 00000000000000000000000 = 0.000030517578
```

We can directly compute these values and convert/transmute them to floating-point values using `f32::from_bits()`:

```rust
/// Value of the max duty cycle field as a portion of time (ex: 0.5).
pub fn max_duty_cycle(&self) -> f32 {
    f32::from_bits(1.0_f32.to_bits() - ((self.max_duty_cycle_raw() as u32) << 23))
}
```

I quite like this approach, although the code is a little difficult to understand. However, the returned value might be difficult to use on systems without floating-point support, so I'm not sure if it resolves this issue.