lora-rs/lora-rs #302: async_device: Integrating mac with "device"

plaes, 2024-12-03:
One of the scenarios current design does not support is `DeviceTimeReq`/`DeviceTimeAns` from LoRaWAN 1.0.3+, which requires following integration:
1. Device can trigger the `DeviceTimeReq` mac frame.
2. Device should be able to receive data from `DeviceTimeAns` mac handler.

For now only following two mac commands seem to be partially implemented/handled in the [async_device](https://github.com/lora-rs/lora-rs/blob/96d2cc25622f43f02328238fcd13166d8d4114f2/lorawan-device/src/mac/mod.rs#L60-L71):
* [ ] LinkADRReq
* [ ] RXTimingSetupReq

And these aren't handled at all:
* [ ] LinkCheckAns
* [ ] DutyCycleReq
* [ ] RXParamSetupReq
* [ ] DevStatusReq
* [ ] NewChannelReq
* [ ] TXParamSetupReq (1.0.2)
* [ ] DlChannelReq (1.0.2)
* [ ] DeviceTimeAns (1.0.3)

lucasgranberg, 2024-12-03:
maybe some inspiration could be joinked from here
https://github.com/lucasgranberg/lorawan/blob/main/src/mac/mod.rs#L394

plaes, 2024-12-04:
# Class A MAC commands (1.0.4)
## Initiated by device
* `LinkCheckReq` (no payload) -> Receive: Noise Margin and Gateway count
* `DeviceTimeReq` (no payload) -> Receive GPS time as 5 bytes (seconds + fractional seconds)
## Initiated by Network
* `LinkADRReq`
  * DataRate -> `mac.configuration.datarate`  **TODO**
  * TXPower -> `mac.board_eirp: BoardEirp { max_power, antenna_gain }` **TODO**
  * Channel Mask -> `mac.region` (NB! Region specific!)
  * NBTrans (**TODO**)
* DutyCycleReq - **TODO**
* RXParamSetupReq -> **Persistency required (class C?)**
  * DLSettings.RX1DROffset -> `mac.region.rx1_offset` **TODO**
  * DLSettings.RX2DataRate -> `mac.region.rx2_dr` **TODO**
  * Frequency (for second data window) **TODO** (no support at all)
* DevStatusReq
  * Battery ->  Need to query underlying hardware (by default return 0 for non-battery device) **TODO**
  * RadioStatus -> Pass `RxQuality`  to `mac.configuration.handle_downlink_macs()` via `mac.handle_rx{,c}` **TODO**
* NewChannelReq - Depends on region whether it should be handled
  * Channel Index - ** TODO **
  * Frequency -  ** TODO **
  * DRRange -  ** TODO **
* DlChannelReq - **Persistency required (class C?)**  - Depends on region whether it should be implemented
* RXTimingSetupReq - **Persistency required (class C?)** - Depends on region whether it should be implemented
  * RX1 delay - `mac.region.rx1_offset`
* TXParamSetupReq  - **Persistency required (class C?)** - Depends on region whether it should be implemented
  * DownlinkDwellTime - **TODO**
  * UplinkDwellTime - **TODO**
  * MaxEIRP - **TODO**

lthiery, 2024-12-24:
My opinions on a few of these topics:

1) @plaes I believe there is quite a finite subset of MAC commands that should be exposed to the user/application and we should separate those out. ADR related MAC commands should all be handled internally, for example.

2) @lucasgranberg we already keep MAC and Radio quite distinct. The top-level `Device` held by the user is as follows:
```rust
pub struct Device<R, C, T, G, const N: usize = 256, const D: usize = 1>
where
    R: radio::PhyRxTx + Timings,
    T: radio::Timer,
    C: CryptoFactory + Default,
    G: RngCore,
{
    crypto: PhantomData<C>,
    radio: R,
    /// Access to provided (pseudo)-random number generator.
    pub rng: G,
    timer: T,
    mac: Mac,
    radio_buffer: RadioBuffer<N>,
    downlink: Vec<Downlink, D>,
    class_c: bool,
}
```
I might be mistaken, but the natural implication of your suggestion is that I have to pass the Device to the Mac struct for every Lorawan operation and I don't like those ergonomics very much. I believe most users will be happy to have a high level structs that owns everything.

However, I do think we should find a way to enable your use-case and I wonder if we simply need a way to decompose and recompose the Device. For example:
```rust
impl From<(Mac, R)> for Device<...> 
impl From<Device<...> for <(Mac, R>>
```
I believe this would fit your use-case while also make it so that a fully assembled Lorawan Device may be maintained. Would that work for you?

3) @lucasgranberg I'm not sure I understand your point about state in regions. 

> Next I would also make Region stateless. I don't recall a single region rule that would require its own state (I have not implemented them all). 

You say yourself that channel plans have state; and as far as I'm concerned, all of the channel/frequency stuff needs to be handled within Region implementations. Insofar as the US915 implementation, there's pretty complex state required to implement our notion of join_bias (which remains within the specification) and to implement "randomness" in a way that suits the US regulatory standards (FCC). So there's a lot of state that is quite regionalized. 

> The benefit of this is that the Region can be made in to a trait and that way it is easier to implement correctly for every region.

How does state prohibit trait definitions? `&self` and `&mut self` on any fn signature allow you to interact with state. Our highest level region trait is defined [here](https://github.com/lora-rs/lora-rs/blob/main/lorawan-device/src/region/mod.rs#L498-L528).

What you might be missing is the fact that we have a layer of abstraction between Fixed and Dynamic channel plans. A layer below that we do have some stateless region definitions which covers frequencies, eg: [EU868](https://github.com/lora-rs/lora-rs/blob/3dac96484d97636c61c667e4c4ff4d80c02b11b0/lorawan-device/src/region/dynamic_channel_plans/eu868.rs). 

And with this pattern we've been able to scale the regional codebase to address 9 regions where addition of regions is fairly painless. Moreover, I hoped that this setup would make it so properly implementing ADR commands would either apply to all regions at once or at the very least, be handed in the Dynamic and Fixed channel plan logic. For the most part, I believe ADR handling requires the region-level to verify the correctness of a settings before accepting it (eg: you've added another channel, but is it within the acceptable channels of this region?).

But anyway, backing up to perhaps your main point: does the current paradigm limit the implementation of something for you or does it cause problems in another way? Because I don't believe I understood your point about state.

lucasgranberg, 2024-12-25:
2. I think the ergonomics of passing the device to Mac is actually more ergonomic, at least the way I use it in a low-power setting. The standard for low power stuff is usually to drop the driver (Device) but we want to keep Mac as there is state that needs to be saved during low power.

The way I use it is: 

- wake up
- get peripherals from embassy init
- measure stuff and do other non-lorawan stuff
- drop all the drivers and get back just the peripherals
- create Device from peripherals
- recreate Mac from flash(if power cycled) or use Mac from SRAM2 (after sleep)
- mac.send(data,device)
- device.drop()
- sleep

Juggling the device struct would not be that hard. I also use "Device" in my code as a way to communicate out of band stuff such as device capabilities (DeviceSpec in my code) to the Mac. if Device owns Mac this becomes harder.

Another place where having Mac borrow device or "be aware" of device is when handling maccommands such as DeviceTimeAns and DevStatusAns.

That being said as long as I can do what I need I can work with any API, preferably one that makes sense to as many as possible.

3. I have to think about more. Looks like I am totally wrong here. Reading just "LoRaWAN® L2 1.0.4 Specification" and implementing EU868 and US915 I got the impression that the data structs could be the same for all regions and the regions would just contain const rules. That way the actual state(dynamic data) could reside in the Mac struct (and its sub structs such as configuration and session). This has worked for me so far but as I said I have not implemented all regions. Also I have not touched class C.

lucasgranberg, 2024-12-27:
looking at 3. again. @lthiery is spot on here:
>But anyway, backing up to perhaps your main point: does the current paradigm limit the implementation of something for you or does it cause problems in another way? Because I don't believe I understood your point about state.

I think I get confused by this bit
https://github.com/lora-rs/lora-rs/blob/3dac96484d97636c61c667e4c4ff4d80c02b11b0/lorawan-device/src/mac/mod.rs#L50-L76

- We pass in something we call region.
- region is a Configuration.
- Configuration only contains "State".
- State is a ChannelPlan.
- ChannelPlan is a FixedChannelPlan or DynamicChannelPlan
- ChannelPlan is Generic over DynamicChannelRegion or FixedChannelRegion

In the end this all boils down to pretty much the same information as the stuff I have and there is no state in region (state as in region specific variables). ie. the region is stateless, just the way I wanted it. This is maybe not the most easy to follow but should not be a problem for implementing all maccommands.

lucasgranberg, 2024-12-27:
Now the blocker as I see it is that we can't access all the information and all the variables we need inside this handle_downlink_macs.

For example LinkADRReq. Here we get DataRate, TXPower and ChMask.

**DataRate** needs to be checked agains the Region to see what the DR values actually means. Here are some RFU so in theory this could fail. I would save this to something like mac.configuration.dr (or self.configuration.dr as maccommand handling should happen inside the Mac context.

**TxPower** needs some validation
>An end-device SHALL acknowledge the successful
receipt of a command that specifies a higher TX power than it is capable of using. In that case,
the end-device SHALL operate at its maximum possible power. An end-device will negatively
acknowledge a command that specifies a lower TX power than the end-device is capable of
using. In that case, the end-device SHALL operate at its previously configured TX power.

So here we need to know the lowest power the device is capable of. I used DeviceSpec trait here. I put this information in mac.configuration.tx_power 

**ChMask** needs access to the ChannelPlan. the channel mask raw data has the same shape regardless of region or type of channel mask (fixed or dynamic). Always 2 bytes.

Here ChannelMask could be a common trait for Fixed or Dynamic and we could pass in these two bytes to validate and set the value. I put this in mac.channel_mask.

This was only one example. afaik this is the information we need to handle the maccommands:

- &mut mac.configuration
- &mut mac.channel_plan
- device specs, I used a trait with fns that return what the device is capable of
- some kind of maccommand handler for maccommands that are handled by the user ex. LinkCheckReq. I used "Device" here just to have only one thing for the user to impl but it could be some other thing too.
- Region constants. I used a Region trait here
- PacketStatus for DevStatusReq

lucasgranberg, 2024-12-27:
> You say yourself that channel plans have state; and as far as I'm concerned, all of the channel/frequency stuff needs to be handled within Region implementations. Insofar as the US915 [...]

I think what you call join_bias is called preferred_join_channel_block_index. This code was written by ceekdee and I don't understand the US915 spec that well. I don't see this as something dictated by the region other than the region spec asking for this information. This is why it lives in "Device" in "my" code.

lthiery, 2025-01-03:
> For example LinkADRReq. Here we get DataRate, TXPower and ChMask.

@lucasgranberg I think that's pretty much the perfect MAC command to use when trying to lay things out to scale, because as you point out, properly handling it requires knowledge of region and board and, therefore, we need our code organization to enable that without too much fuss.

For the sake of discussion, I'll start from the top-level struct in lora-rs, `Mac`:
```rust
pub(crate) struct Mac {
    pub configuration: Configuration,
    pub region: region::Configuration,
    board_eirp: BoardEirp,
    state: State,
}
```
It contains all of the "extra" components needed to properly handle LinkADRReq, ie: `region` and `board_eirp`. Side note, if we can implement an efficient serializer/deserializer for this entire struct, than I believe we will be a step closer to what you need for your low power state.

Anyway, this might be a mistake, but currently, we pass all MAC commands to [`mac::Configuration` here](https://github.com/lora-rs/lora-rs/blob/main/lorawan-device/src/mac/mod.rs#L40-L76) for handling; I believe I modeled Configuration here after [your configuration](https://github.com/lucasgranberg/lorawan/blob/main/src/mac/types.rs#L33-L46). Perhaps it would make more sense to allow the top level Mac struct to delegate to all the pieces depending on command - I'm not sure.

However, as you can see, we already passing `region::Configuration` and our minimal LinkADRReq implementation passes it to region. The first mistake here is that there's no opportunity for the region layer to reject this setting and we exclusively ACK on the next line with `uplink.adr_ans.add()`. The second mistake, is that we don't do anything with the transmit power and that should also be validated with a combination of region and board_eirp.

That being said, I believe all the pieces are in place for implementation?

lucasgranberg, 2025-01-03:
> Side note, if we can implement an efficient serializer/deserializer for this entire struct, than I believe we will be a step closer to what you need for your low power state.

One thing to note is that what is saved during sleep and what is saved during power cycle is different. I do re-join on power loss as I have no idea of what happened in between. For power loss I persist [this](https://github.com/lucasgranberg/lorawan/blob/3c0f01d30a01d9802f1e47a2f32de13b61eabadb/src/mac/types.rs#L152) to flash

> Perhaps it would make more sense to allow the top level Mac struct to delegate to all the pieces depending on command - I'm not sure.

yes, if we have handle_downlink_macs in Mac we should have all the information needed. I think there is still some "out of band" stuff that is needed but that can be added as needed.

> The first mistake here is that there's no opportunity for the region layer to reject this setting and we exclusively ACK on the next line with uplink.adr_ans.add(). The second mistake, is that we [...]

The way I did it was to validate in region and then add to the Mac state or configuration (not sure about the terms here). In my Mac I have a lot of these kinds of fns
```rs
    fn max_eirp<D: DeviceSpecs>() -> i8 {
        if let Some(device_max_eirp) = D::max_eirp() {
            min(R::max_eirp(), device_max_eirp)
        } else {
            R::max_eirp()
        }
    }
 ```
 
 The actual value could be put in to the configuration directly but I left it like this as it was easier to debug and see why the final value was chosen. Here we have configuration via maccommands, device capabilities and region "fighting" over the final value. The spec is pretty clear on how this should be selected.
 
 > That being said, I believe all the pieces are in place for implementation?
 Yes, I guess first step would be to move handle_downlink_macs to the Mac level?

lthiery, 2025-01-04:
> One thing to note is that what is saved during sleep and what is saved during power cycle is different. I do re-join on power loss as I have no idea of what happened in between. For power loss I [...]

That's a very good point. Many of those "Storable" fields are out-of-band configuration options typical of LoRaWAN devices. That is a different category than saving and restoring the Mac sessions from a deep sleep.

> Yes, I guess first step would be to move handle_downlink_macs to the Mac level?

Yes, I think moving the current hack features to be at the Mac scope would be an easy PR for a step in the right direction. Maybe even a proper implementation of LinkADRReq. From there, I think we will have the framework setup and building out additional commands will be a matter of putting in the work.

plaes, 2025-02-03:
> Yes, I think moving the current hack features to be at the Mac scope would be an easy PR for a step in the right direction. Maybe even a proper implementation of LinkADRReq. From there, I think we [...]

Managed to to this with #332

My next blocker right now is implementing the Certification protocol's DutResetReq and DutJoinRec.

plaes, 2026-11-11:
Initial DeviceHandler implementation is now in #511