lora-rs/lora-rs #312: Sx127x IRQ Confusion

martin2250, 2024-12-28:
Hi,

I'm trying to make an RFM92 Module (Sx1276) work on my custom board, controlled by an stm32l053r8 with embassy. However, the application seems to lock up every time it expects to receive a packet and no packet arrives.

## What I Did

I used the [STM32L0 example](https://github.com/lora-rs/lora-rs/blob/main/examples/stm32l0/src/bin/lora_lorawan.rs), only changing the device keys, region, the pin mapping and adding a loop to actually send data at the end of the main function:
```rust
    // only for example, I tried all DIOs. used in the `GenericSx127xInterfaceVariant`
    let dio0_pin = Input::new(p.PB1.degrade(), Pull::Up);
    let dio0_irq = ExtiInput::new(dio0_pin, p.EXTI1.degrade());
    // at the end of main, after the join
    loop {
        device.send(&[0xde, 0xad, 0xbe, 0xef], 1, false).await.unwrap();
        Timer::after(Duration::from_secs(30)).await;
    }
```
For the IRQ pin I wasn't sure which DIO to use (my board exposes all to the STM32), so I tried all of them. For some, the transmit didn't succeed. If that worked, the application would get completely stuck:
- when the join accept message wasn't received
- when nothing was received during the first RX window (even unconfirmed messages)

## Debugging

I believe that the issue is that with the current way IRQs are handled by the SX127x implementation, the timeout interrupt is not received. In `set_irq_params`, the mask is manipulated for both the DIO0 and DIO3 pins. We're only able to configure a single pin, however.

I got everything to work perfectly by spoofing the irq pin like this:
```rust
struct Irq;

impl embedded_hal::digital::ErrorType for Irq {
    type Error = core::convert::Infallible;
}

impl embedded_hal_async::digital::Wait for Irq {
    async fn wait_for_high(&mut self) -> Result<(), Self::Error> {
        Timer::after_millis(100).await;
        Ok(())
    }

    async fn wait_for_low(&mut self) -> Result<(), Self::Error> {
        defmt::todo!()
    }

    async fn wait_for_rising_edge(&mut self) -> Result<(), Self::Error> {
        defmt::todo!()
    }

    async fn wait_for_falling_edge(&mut self) -> Result<(), Self::Error> {
        defmt::todo!()
    }

    async fn wait_for_any_edge(&mut self) -> Result<(), Self::Error> {
        defmt::todo!()
    }
}
```
This makes the driver effectively use polling (this might be interesting for #252). Since this works well, I believe that the hardware and dev environment are working.

I'm using lora-rs from git on commit [3dac964](https://github.com/lora-rs/lora-rs/commit/3dac96484d97636c61c667e4c4ff4d80c02b11b0) and embassy from crates.io
```toml
embassy-stm32 = { version = "0.1.0", features = [
    "stm32l053r8",
    "time-driver-any",
    "memory-x",
    "exti",
    "unstable-pac",
] }
```

plaes, 2024-12-29:
You need both dio0 and dio1 for sx1276 because lorawan-device uses `RxTimeout` IRQ:
[image]

See pseudocode from this example:
 https://github.com/lora-rs/lora-rs/issues/262#issuecomment-2476016424

The example is broken because none of the currently active developers have this device and some time ago, lorawan-device was reworked to handle RX2 windows using hardware timeout (which relies on RxTimeout IRQ).

martin2250, 2024-12-30:
seems to me that this applies to all 127x chips, then we could just change `GenericSx127xInterfaceVariant` to take `dio0` and `dio1` instead of `irq`, right? I can create a PR, but I don't have the board here anymore to test it.

plaes, 2024-12-30:
Yeah it's a bit tricky with examples because p2p works fine with only dio0. Probably best one can do is go over the all examples and mark the ones which use lorawan as broken for now.
I have previously created issues regarding this mess, but haven't yet had time to properly sort it out:
* #238 
* #239

Kheirlb, 2025-08-25:
+1 for updated example

tsar-boomba, 2025-09-25:
I'm gonna implement `dio0` and `dio1` for `GenericSx127xInterfaceVariant` since I need it for a school project, and I have the hardware. PR should be up before the end of the semester.