lora-rs/lora-rs #313: Make Class-C optional?

martin2250, 2024-12-30:
Hi, would you be interested in (aka accept a pull request) putting the LoRaWAN Class-C features behind a (default) feature flag?

I'm using lora-rs on a rather memory-constrained device (64K STM32) and the class-c code alone uses about ~10kB (according to cargo-bloat, specifically one closure in `between_windows`. By removing all class-C code (and panic-probe plus some others), I just managed to fit it in.

It would be nice to have a feature flag that could be used instead.

Also, in the same spirit, a transmit-only feature could be useful for ABP devices. I'm guessing, however, that this would require feature-gating a lot more sections of code.

plaes, 2024-12-30:
Yes, when disabling Class C drops ~10kB, adding a feature flag for it makes sense.

Also, not against the transmit-only feature, but our current priority is getting MAC+application layer properly work together to support things required for full 1.0.4 support (ADR, channel and timing management, etc see #302).

martin2250, 2024-12-30:
nvm, while working on the PR I found that the 10kB drop was not caused by class-c. I'm guessing what I saw was only panic-probe and defmt features in other crates.

The actual difference with / without class-c is only around 1kB and therefore probably not worth the complexity.

Cheers!

closed 2024-12-30 by martin2250

lthiery, 2025-01-03:
I was going to say, I was surprised that Class C would add such overhead since it is mostly a procedural change in listening and times when otherwise we'd been sleeping! I'd be open to feature-gating it though as I am working on a feature-gated multicast setup as well.

martin2250, 2025-01-04:
I still have the code in my fork (not 100% ready), I'll go over it later and open a PR if you'd still take it.