lora-rs/lora-rs #316: Simplify Crypto and RNG generics with defaults

lthiery, 2025-01-03:
I believe we could make use of our default implementations of Crypto and RNG to simplify usage of our APIs.

For example, the async device is currently defined as follows:
```rust
pub struct Device<R, C, T, G, const N: usize = 256, const D: usize = 1>
where
    R: radio::PhyRxTx + Timings,
    T: radio::Timer,
    C: CryptoFactory + Default,
    G: RngCore,
```
I believe we could reshuffle the order and add more defaults at the end here:
```
pub struct Device<R, T, C = DefaultCrypto, G = PRNG, const N: usize = 256, const D: usize = 1>
```

This same idea applies to the lorawan encoding crate and `CryptoFactory` but I'm unsure if we can avoid passing it as a fn generic parameter.

lthiery, 2025-01-03:
@lucasgranberg reminded me that DefaultCrypto is feature-gated here. I wonder if it's worth not feature-gating it... is it worth including it in the build always to simplify the happy path for a device definition?

plaes, 2025-05-19:
I guess we can close this and #347 as well now?

lucasgranberg, 2025-05-19:
We can close this but maybe leave #347 for now. The way I did it in https://github.com/lora-rs/lora-rs/pull/382 still leaves some mess behind but at least the Device API is simplified a bit. The mess being that we have a crypto trait but now way of using your own crypto impl with the current lorawan-device code.

closed 2025-05-19 by lucasgranberg