lora-rs/lora-rs #347: overhaul crypto interfaces?

lthiery, 2025-02-14:
I've been thinking about the crypto interfaces and considered at the very least removing the generic from lorawan-device (#316).

However, I think it might be worth removing it entirely as a generic in lorawan-encoding. I'm not sure the flexibility is particularly useful.

Moreover, while doing this, I think we should make better use of instantiating encryption/decryption blocks only a single time and re-using them for future operations. [This](https://github.com/lora-rs/lora-rs/blob/f9c5c4ee98a6cf0fdeb60d65211d59b0e44c3e09/lorawan-encoding/src/creator.rs#L547) type of thing for example is a bit wasteful. It's only 16 bytes of more RAM usage but would probably save quite a few operations if we kept the same block around.

Thoughts?

plaes, 2025-02-14:
I'm all for it. Besides, it seems we'll be gaining a new generic anyway which contains "user defined" device functionality to support Device time and DeviceReq MAC commands. It might be possible to use this later for crypto as well.

lucasgranberg, 2025-02-14:
I am all for it. One thing to think about is that do we really need an JoinAcceptCreator in the device code? The crypto stuff has both encrypt and decrypt to suite both network and device. The spec is made in such a way that only encrypt is needed on the device (even for decrypting messages).

lucasgranberg, 2025-05-07:
Instead of having a cryptofactory like this:
https://github.com/lora-rs/lora-rs/blob/f9c5c4ee98a6cf0fdeb60d65211d59b0e44c3e09/lorawan-encoding/src/default_crypto.rs#L18-L35

We should probably have something more simple like
```rs
trait Crypto{
    type Error;
    fn calculate_mic(buf: &[u8]) -> [u8; 4];
    fn encrypt(buf: &mut [u8]);
}
```

Creating a new_enc or new_cmac every time is probably not a big problem but should be up to the impl to decide. For soft cmac it needs to be reset between mic calculations anyway.

I agree that we should remove the CryptoFactory generic from Device but as @plaes said about the "user defined" stuff it needs to come in somewhere.

Sorry for being a broken record but this all circles back to my gripes with what should live where.

Basically I want to do
```rs
mac.join(&mut device) //pass in keys here or better yet let the device handle the keys
mac.send(&mut device,data)
```
where lora-rs currently does
```rs
device.send(data)
device.join(keys)
```

Doing it "my way" makes mac more purely just the stuff in lorawan specs and makes it easier to drop the drivers and just keep what needs to be saved during sleep.

Getting back to the original issue, doing it like this would put the crypto stuff in device and could be implemented in hw or sw. Also it would allow for keeping the keys only in hw (session keys would still live in mac).

The way Device has CryptoFactory as a generic is kind of messed up right now as only a sw crypto is possible anyway as we just depend on Default::default() for it.

lucasgranberg, 2025-05-07:
https://github.com/lora-rs/lora-rs/compare/main...lucasgranberg:lora-rs:nuke-crypto-generic

I gave removing crypto generic from device a go. I can't create a draft PR for some reason. maybe it doesn't like my branch name or something.