lora-rs/lora-rs #364: DevEUI bytes inverted while trying to join on OTAA

JJaviMS, 2025-03-11:
Hi, I am getting started using the lorawan-device. I have found out (after a lot of time not seeing the error) that the DevEUI bytes are sent in wrong order (at least on the RP Pico).

I have the following code: 

```rust
let mut dev = [0x70, 0xB3, 0xD5, 0x7E, 0xD8, 0x00, 0x41, 0x52];
dev.reverse(); // Reorder the bytes so they arrive in the correct order

let resp = device
        .join(&JoinMode::OTAA {
            deveui: DevEui::from(dev),
            appeui: AppEui::from([0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]),
            appkey: AppKey::from(/*My app key*/),
        })
        .await
        .unwrap();
```

And as you can see on this screenshot from the GW package debugger the DevEUI has arrived on the correct order.

[image]

Best regards.

plaes, 2025-03-12:
Unfortunately this is how it's specified in standard, all EUIs are specified as least significant bit (LSB) first, while keys are defined as MSB.

This requirements seems to come from IEEE standard which defines how EUI, OUI and CID are used:
> Typically, octet-based transmissions transmit in ascending order of octet
identifiers, beginning with octet 0.

Keys are used as AES-128 "type", so this is where the MSB byte order might come from...

And we just follow the LoRaWAN standard.

If you have ideas how to improve the documentation, we'll be glad to hear from you:
https://docs.rs/lorawan/latest/lorawan/keys/struct.DevEui.html

JJaviMS, 2025-03-14:
Hi, thanks for the answer.

I would suggest two improvents that may work:

The first one is add this information also on the example, so people that goes directly to the example can see this information.

The second suggestion is to add a new method that allows to directly reverse the bytes when creating the DevEui or the AppKey, I think something like this could fit:

```rust
trait FromMsbToLsb: From<[u8; 8]> {
    fn new_from_msb(mut source: [u8;8]) -> Self {
        source.reverse();
        Self::from(source)
    }
}
impl FromMsbToLsb for DevEui {}
```

Maybe the implementation or naming could be improved, but the main idea is to provide a method that performs the conversion for you.